openwrt/packages #1338: redsocks compile error

taeasy, 2015-06-01:
mv -f gen/version.c.tmp gen/version.c
gcc -MM parser.c main.c redsocks.c log.c http-connect.c socks4.c socks5.c http-relay.c base.c base64.c md5.c http-auth.c utils.c redudp.c dnstc.c gen/version.c 2>/dev/null >.depend || \
	( \
		for I in base.h base64.h dnstc.h http-auth.h libc-compat.h libevent-compat.h list.h log.h main.h md5.h parser.h redsocks.h redudp.h socks5.h utils.h version.h; do \
			export ${I//[-.]/_}_DEPS="`sed '/^\#[ \t]*include \?"\(.*\)".*/!d;s//\1/' $I`"; \
		done; \
		echo -n >.depend; \
		for SRC in parser.c main.c redsocks.c log.c http-connect.c socks4.c socks5.c http-relay.c base.c base64.c md5.c http-auth.c utils.c redudp.c dnstc.c gen/version.c; do \
			echo -n "${SRC%.c}.o: " >>.depend; \
			export SRC_DEPS="`sed '/\#[ \t]*include \?"\(.*\)".*/!d;s//\1/' $SRC | sort`"; \
			while true; do \
				export SRC_DEPS_OLD="$SRC_DEPS"; \
				export SRC_DEEP_DEPS=""; \
				for HDR in $SRC_DEPS; do \
					eval export SRC_DEEP_DEPS="\"$SRC_DEEP_DEPS \$${HDR//[-.]/_}_DEPS\""; \
				done; \
				export SRC_DEPS="`echo $SRC_DEPS $SRC_DEEP_DEPS | sed 's/  */\n/g' | sort -u`"; \
				test "$SRC_DEPS" = "$SRC_DEPS_OLD" && break; \
			done; \
			echo $SRC $SRC_DEPS >>.depend; \
		done; \
	)
make[4]: Leaving directory `/home/main/n16/build_dir/target-mipsel_74kc+dsp2_uClibc-0.9.33.2/redsocks-0.4'
make[4]: Entering directory `/home/main/n16/build_dir/target-mipsel_74kc+dsp2_uClibc-0.9.33.2/redsocks-0.4'
mipsel-openwrt-linux-uclibc-gcc -Os -pipe -mno-branch-likely -mips32r2 -mtune=74kc -mdspr2 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16  -I/home/main/n16/staging_dir/target-mipsel_74kc+dsp2_uClibc-0.9.33.2/usr/include -I/home/main/n16/staging_dir/target-mipsel_74kc+dsp2_uClibc-0.9.33.2/include -I/home/main/n16/staging_dir/toolchain-mipsel_74kc+dsp2_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/main/n16/staging_dir/toolchain-mipsel_74kc+dsp2_gcc-4.8-linaro_uClibc-0.9.33.2/include  -g -O2 -std=gnu99 -Wall   -c -o parser.o parser.c
In file included from parser.c:29:0:
utils.h:43:41: warning: 'struct evbuffer' declared inside parameter list [enabled by default]
 char *redsocks_evbuffer_readline(struct evbuffer *buf);
                                         ^
utils.h:43:41: warning: its scope is only this definition or declaration, which is probably not what you want [enabled by default]
utils.h:44:65: error: unknown type name 'evbuffercb'
 struct bufferevent* red_connect_relay(struct sockaddr_in *addr, evbuffercb writecb, everrorcb errorcb, void *cbarg);
                                                                 ^
utils.h:44:85: error: unknown type name 'everrorcb'
 struct bufferevent* red_connect_relay(struct sockaddr_in *addr, evbuffercb writecb, everrorcb errorcb, void *cbarg);
                                                                                     ^
make[4]: *** [parser.o] Error 1
make[4]: Leaving directory `/home/main/n16/build_dir/target-mipsel_74kc+dsp2_uClibc-0.9.33.2/redsocks-0.4'
make[3]: *** [/home/main/n16/build_dir/target-mipsel_74kc+dsp2_uClibc-0.9.33.2/redsocks-0.4/.built] Error 2
make[3]: Leaving directory `/home/main/n16/feeds/packages/net/redsocks'
make[2]: *** [package/feeds/packages/redsocks/compile] Error 2
make[2]: Leaving directory `/home/main/n16'
make[1]: *** [/home/main/n16/staging_dir/target-mipsel_74kc+dsp2_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/main/n16'
make: *** [world] Error 2
main@bogon:~/n16$

same problem also in buildbot: 

http://buildbot.openwrt.org:8010/broken_packages/ar71xx/redsocks/compile.txt

gestapolur, 2015-09-01:
Probably caused by missing libevent2 header. but this lib should already in redsocks's Makefile: https://github.com/darkk/redsocks/blob/master/Makefile#L8 Same problem in this issue: https://github.com/semigodking/redsocks/issues/23 

confusing.

qiuchengxuan, 2015-10-12:
I succeeded compile this with redsocks installed only

alzhao, 2015-11-05:
The reason is, in OpenWrt CC, libevent2 is upgraded to 2.0.12. If you downgrade to 2.0.11, autossh will compile without problem. Maybe need to write a patch.

thess, 2015-11-15:
I have tried to re-create this problem (same issue on buildbots) but have been unable to. I would like someone who does have this problem to test PR #1965. TIA

thess, 2015-11-28:
Problem is not resolved with latest compiler options. It would be nice to know that the contents of .../usr/include/event.h (in staging_dir/target_xxx) contains on a system where this fails.

thess, 2015-12-01:
OK - I found the problem. There is a conflict between libev and libevent2. Libev overwrites the libevent2 include files (event.h in particular). I have no quick fix short of disabling libev builds. @karlp - any suggestions? There doesn't seem to be any other package dependent on libev?

PR #2112 (2015-12-30): libev: Fix compatibility with libevent2 includes
Packages building with libev and want to include event.h from libev
need to be modified to include ev_event_compat.h instead.

The are no known packages which need to be fixed at this time. The
normal include is ev.h for this package.

Does not break Kamailio4 and also fixes #1338
closed 2015-12-30 by thess