openwrt/packages #1341: [Luci] Missing WPA/WPA2-EAP config support with just wpa-supplicant+hostapd installed

hutamanzi, 2015-06-03:
Hi,
I hope I submitted this bug on the right place according to wiki http://wiki.openwrt.org/doc/devel/bugs

The radio was configured as client of my organisation wlan with wpa2-enterprise encryption
If I installed wpad without wpa-supplicant and hostapd, I could see the WPA2-EAP option in encryption configuration luci. But if I just uninstall wpad and replace it with wpa-supplicant and hostapd, luci doesn't show the WPA2-EAP option. I should then either configure it through terminal and edit etc/config/wireless or install wpad, configure it through luci and then uninstall wpad again and replace with wpa-supplicant and hostapd. Wpad doesn't support wpa2-eap 5ghz ath10k very well at this moment, that's why I use wpa-supplicant and hostapd rather than wpad.

fyi: it still exists not only in bb but also in the newest luci trunk chaos calmer
[image]

diizzyy, 2017-06-07:
@thess @hnyman 
Close, wrong repo.

closed 2017-06-07 by thess

hnyman, 2017-06-07:
Adding a note about issue:
core reason is the limited feature support indication in the old hostapd/wpad in Openwrt. 

Better feature indication has been added to hostapd in LEDE master with https://github.com/lede-project/source/commit/aff2d5c856f34f46442f83974d4969c33ab1c563 and also in LEDE 17.01, but no fix has been made to Openwrt repo.