openwrt/packages #1343: mdnsresponder starts with sockopt errors

mehlis, 2015-06-03:
```
$ mdnsd -debug
mDNSResponder (Engineering Build) (May 28 2015 14:44:28) starting
  7: Listening for incoming Unix Domain Socket client requests
mDNS_AddDNSServer: Lock not held! mDNS_busy (0) mDNS_reentrancy (0)
setsockopt - SO_RECV_ANYIF: Protocol not available
setsockopt - SO_RECV_ANYIF: Protocol not available
setsockopt - SO_RECV_ANYIF: Protocol not available
setsockopt - SO_RECV_ANYIF: Protocol not available
setsockopt - SO_RECV_ANYIF: Protocol not available
setsockopt - SO_RECV_ANYIF: Protocol not available
setsockopt - SO_RECV_ANYIF: Protocol not available
setsockopt - SO_RECV_ANYIF: Protocol not available
setsockopt - SO_RECV_ANYIF: Protocol not available
setsockopt - SO_RECV_ANYIF: Protocol not available
setsockopt - SO_RECV_ANYIF: Protocol not available
setsockopt - SO_RECV_ANYIF: Protocol not available
setsockopt - SO_RECV_ANYIF: Protocol not available
setsockopt - SO_RECV_ANYIF: Protocol not available
setsockopt - SO_RECV_ANYIF: Protocol not available
```

sbyx, 2015-06-10:
Does this actually break anything or is this just cosmetical?

diizzyy, 2017-06-07:
@hnyman @thess 
Close, reporter timeout.

closed 2017-06-07 by hnyman