openwrt/packages #1351: update collectd to 5.5.0 and make Write Graphite optional

diizzyy, 2015-06-06:
The graphite plugin works fine on the current port, not sure why it's disabled/not optional.

hnyman, 2015-06-23:
I tested bumping the collectd version from 5.4.2 to 5.5.0, and the following patch seems to compile and work both in CC15.05 (with uclibc) and in trunk (with musl).
https://gist.github.com/hnyman/903c73938cc56fd9218f

I did not yet test compiling the write-graphite plugin that you mentioned. When you tried it, did you need any other changes in Makefile than moving the plugin from the "disabled" group to the "enabled" group and adding a line to the plugin definition list at the end?

I might later create a pull request for trunk, but as dependencies of some collectd plugins still have musl problems, there is currently no way to compile-test all plugins.

diizzyy, 2015-06-30:
Hi, 
Terribly sorry for the late response... :/
As for the plugin I just moved it around in the Makefile and that worked fine with musl.
For now, I think the current list makes sense.

Thanks for the work!

karlp, 2015-07-19:
this is merged, believe we can close this now...

closed 2015-07-19 by karlp