openwrt/packages #1353: vpnc can't work.

benjaminlei2007, 2015-06-08:
I installed the vpnc and vpnc-scripts packages.
But there's no scripts in /etc/vpnc.
I manually upload the scripts, and make sure the default.conf is correct, but it still can't work.
I use the 'ifconfig' command, and it shows no tun devices.
Is there anybody can fix it? Thank you!

nmav, 2015-06-25:
@danielg4

danielg4, 2015-06-25:
Can you show the contents of /etc/config/network, obfuscating your VPN credentials?

benjaminlei2007, 2015-06-26:
/etc/config/network
                               
config interface 'loopback'                    
        option ifname 'lo'                     
        option proto 'static'                  
        option ipaddr '127.0.0.1'              
        option netmask '255.0.0.0'             
                                               
config globals 'globals'                       
        option ula_prefix 'fd3e:8d7a:2dd8::/48'
                                      
config interface 'lan'                
        option ifname 'eth0.1'        
        option force_link '1'         
        option type 'bridge'          
        option proto 'static'         
        option netmask '255.255.255.0'
        option ip6assign '60'      
        option ipaddr '192.168.1.2'
                               
config interface 'wan'         
        option ifname 'eth0.2'   
        option proto 'dhcp'      
                                 
config interface 'wan6'          
        option ifname 'eth0.2'   
        option proto 'dhcpv6' 

config switch                         
        option name 'switch0'         
        option reset '1'              
        option enable_vlan '1'        
                                   
config switch_vlan                 
        option device 'switch0'
        option vlan '1'        
        option ports '0t 2 3 4 5'
                                 
config switch_vlan               
        option device 'switch0'  
        option vlan '2'          
        option ports '0t 1'

danielg4, 2015-06-26:
You don't seem to have any entries for the VPN in the file. Did you not read the package README file at all?

benjaminlei2007, 2015-06-26:
I used to install the vpnc and vpnc-scriptis via 'opkg' on Barrier Breaker (14.07). It seemed no need to add the entries for vpn. I just configured  the VPN server and credentials in /etc/vpnc/default.conf, and modified the /etc/vpnc/vpnc-scrpit to enable NAT though VPN. And I created a startup script to invoke 'vpnc' to initial connection. It works very well on Barrier Breaker (14.07). 
But the same way can't work on Chaos Calmer(trunk).
I'll try the way the README mentioned. 
Thank you!

danielg4, 2015-06-26:
The new package does not use the old scripts. The old package went unmaintained and was removed.

diapluscn, 2015-10-11:
I have added content to /etc/config/network as README and seeing  "option proto 'vpnc'  " in luci. How to fix this.

rafus15, 2015-11-27:
Hi, i'm having the same trouble.
I used to install vpnc and vpnc-scripts and the vpn was working great.
starting with chaos calmer, i know that this went unmaintained and i have to configure it through /etc/config/network but when i try to configure-it with option proto vpnc thinks doesn't seem to work. And indeed it does not connect to the vpn.
Thanks in advance

danielg4, 2015-11-28:
@diapluscn Nobody has written a LuCI module for the new package so far.
@rafus15 Assuming you followed the README, can you show the contents of /etc/config/network, obfuscating your VPN credentials?

tinkercnc, 2016-06-01:
Hello,
I've the same issues.
Installed Chaos Calmer on tl-wr1043nd and vpnc with luci.
But where should this README be.
I can't find any README related to vpnc.

danielg4, 2016-06-01:
$ ls feeds/packages/net/vpnc
Config.in  files  Makefile  patches  README

tinkercnc, 2016-06-01:
You mean the source pkg?

danielg4, 2016-06-01:
Yes. OpenWrt and LEDE are in the category of source-based distro.

tinkercnc, 2016-06-01:
Does this mean, I've to download the whole source?
Just for configure the vpnc client?

danielg4, 2016-06-01:
No. If you're building from source, you already have the source downloaded. If you simply installed using opkg, it's right here on GitHub: https://github.com/openwrt/packages/tree/master/net/vpnc

tinkercnc, 2016-06-01:
A link here 
   https://wiki.openwrt.org/doc/howto/vpn.client.vpnc
to the README in github would be great. ;)

tinkercnc, 2016-06-01:
Yes, my comment above about downloading the whole source was not totally serious.

tinkercnc, 2016-06-01:
But thanks for the response.

danielg4, 2016-06-01:
It would be better if there were a page that instructs in detail the way that one does, instead of just listing the UCI options. Perhaps after all the docs get rewritten, it will exist. Maybe someone can even write a LuCI module, but I doubt there would be enough call unless there is a new major release upstream supporting decent encryption like the servers do.

tinkercnc, 2016-06-01:
Maybe if I have managed to configure this damn thing... ;)

TaiSHiNet, 2016-06-28:
I've been trying as well, but I receive absolutely no output whatsoever when running ifup

rafus15, 2016-06-28:
Hi!
I ended up making it sort of working.
Like most of you I had no stdout when typing ifup vpn.

I found that I had to read the system logs (logread-f command) , where I saw what was going wrong.

My problem then was that vpnc was unable to set the default route right, so in the boot process it would mess up with the routes and I was disconnected from the net even.

The solution was to add "option defaultroute 0 " line in the network config file.
This way I was not losing the connection , and I was establishing the tunnel! 

My problem now is that I can go through the tunnel actually.

I will report back with my findings.
I hope my experience can help someone!

rafus15, 2016-06-28:
Hi!
I ended up making it sort of work.
Like most of you I had no stdout when typing ifup vpn.

I found that I had to read the system logs (logread -f command) , where I saw what was going wrong.

My problem then was that vpnc was unable to set the default route right, so in the boot process it would mess up with the routes and I was disconnected from the net even.

The solution was to add "option defaultroute 0 " line in the network config file.
This way I was not losing the connection , and I was establishing the tunnel! 

My problem now is that I can't  go through the tunnel actually.

I will report back with my findings.
I hope my experience can help someone!

TaiSHiNet, 2016-06-28:
@rafus15 I've been reading logread the entire time, no output at all, could you copy/paste your /etc/config/network ?

Bests

rafus15, 2016-06-29:
Have you checked if an interface is created when you type "ifup vpn"?
I'm not able to paste it right now because the device is not here with me, i'll paste it as soon as i can.
in any case, i just followed the readme file without any changes besides the "option defaultroute 0" addition.

i hope tomorrow i will be able to paste it, sorry about the delay! ;)

rafus15, 2016-07-19:
Hi! sorry @TaiSHiNet for the delay, so much work !

here is the configuration for the vpn connection that i have (obviously the credentials and server name are fake :) ).

This manages to establish the tunnel, but the routes doesn't get added to the table so i'm not going through the tunnel right now. Adding manually the routes would be the fix, but i wasn't able to do it right.

Hope this helps someone!

 
config interface 'vpn'
        option proto 'vpnc'
        option interface 'wan'
        option server 'server'
        option username 'user'
        option password 'pass'
        option authgroup 'auth'
        option passgroup 'passgroup'
        option defaultroute '0'

If I set defaultroute '1' the routes will get all messed up and i get  "network unreachable" errors all over the place.

danielg4, 2016-07-19:
Can "option target_network ..." be applicable to you at all?

rafus15, 2016-07-19:
hi @danielg4 i think it would be the solution, but i'm using it to connect to my unviersity vpn server which assigns different gateways through time so i think it's not applicable for me.

Though i have to research more, because i think i could make it work.

danielg4, 2016-07-19:
AFAICT, you give that option the IP range of the hosts on your university's inside network which you intend to access.

TaiSHiNet, 2016-07-19:
My issue was FAR easier to solve:
I required a reboot after installing vpnc* packages

Sorry I hijacked your thread

danielg4, 2016-07-19:
IMHO, you still contributed to support, @TaiSHiNet.

rafus15, 2016-07-19:
@TaiSHiNet you helped too, don't worry pal!
@danielg4 i will try to do this and I'll report back! 
Thanks for the idea!

TaiSHiNet, 2016-07-19:
Oh and about target_network, beware, that led me to some STRANGE loops.
IIRC it means to tell the other end which network you are **on** 
After you connect, can you show us your routes?

davinunes, 2017-10-17:
"Unsupported protocol type."
In Interface, but i've installed vpnc as well

danielg4, 2017-10-17:
If you mean the web interface, nobody has made a vpnc module for it yet AFAIK. You'll have to edit /etc/config/network manually. @davinunes

davinunes, 2017-10-17:
I should be doing something wrong then, because I edit the file, but the interface does not go up...
config interface 'vpn'
option proto 'vpnc'
option interface 'wan'
option server 'xxxxxxxxx'
option username 'xxxxxx'
option password 'xxxxxxxx'
option authgroup 'VPN1'
option passgroup 'xxxxx'
option defaultroute '0'

danielg4, 2017-10-17:
Are there any clues in the output of the "logread" command?

davinunes, 2017-10-18:
Wed Oct 18 11:36:09 2017 user.notice vpnc: bringing down vpnc
Wed Oct 18 11:36:09 2017 daemon.notice netifd: Interface 'vpn' is now down
Wed Oct 18 11:36:09 2017 daemon.notice netifd: Interface 'vpn' is setting up now
Wed Oct 18 11:36:09 2017 user.notice vpnc: initializing...
Wed Oct 18 11:36:09 2017 user.notice vpnc: executing 'vpnc --no-detach --pid-file /var/run/vpnc-vpn.pid --ifname vpn-vpn --non-inter --script /lib/netifd/vpnc-script /var/etc/vpnc-vpn.conf'
Wed Oct 18 11:36:24 2017 daemon.notice netifd: vpn (2052): /usr/sbin/vpnc: no response from target

This log is strange because the same data allows me to connect in another Linux distribution

davinunes, 2017-10-18:
```
Wed Oct 18 11:42:56 2017 user.notice vpnc: bringing down vpnc
Wed Oct 18 11:42:56 2017 daemon.notice netifd: Interface 'vpn' is now down
Wed Oct 18 11:42:56 2017 daemon.notice netifd: Interface 'vpn' is setting up now
Wed Oct 18 11:42:56 2017 user.notice vpnc: initializing...
Wed Oct 18 11:42:56 2017 user.notice vpnc: executing 'vpnc --no-detach --pid-fil                                                e /var/run/vpnc-vpn.pid --ifname vpn-vpn --non-inter --script /lib/netifd/vpnc-s                                                cript /var/etc/vpnc-vpn.conf'
Wed Oct 18 11:42:57 2017 daemon.notice netifd: vpn (15634): /usr/sbin/vpnc: serv                                                er requested domain, but none set (use "Domain ..." in config or --domain

```
I realized that it was something in the virtualbox's NAT, so I configured the WAN interface in BRIDGE. And I got this new error. How can I pass the blank domain and in the configurations I have already added the line: 

`option domain ''`

danielg4, 2017-10-18:
I'm not sure it's possible to pass that with the code as it is right now, but you could try "''" (single inside double). However, before going down that road, try passing an empty string on the command line (using '') to a manually started /usr/sbin/vpnc instance to check whether the server will accept it.

hscadore, 2018-06-12:
Hello everyone . I'm trying to set up a VPN between openwrt and a paloalto gateway. It works on linux but I'm trying to figure out how it works on openwrt. First I can authenticate to paloalto using default.conf and it shows tun0 when I run ifconfig -a, I can ifup tun0 but I won't get an IP although paloalto shows the authenticated client (openwrt) has an IP associated to it. That said I learned that I need to configure /etc/conf/network, ok I did that but it seems vpnc will only run if I configure /etc/vpnc/default.conf. If I don't configure default.conf vpnc will ask me for the VPN gateway IP.  Should I configure default.conf even with /etc/config/network configured? I read somewhere that vpnc will pass the IP via env variables is that true? Is that the way I will know the IP I have tos et to tun0? Sorry for so many questions but I'm sort of lost.

danielg4, 2018-06-13:
There should be no need to touch default.conf if you set
/etc/config/network up as in the
https://github.com/openwrt/packages/blob/master/net/vpnc/README
directions.

hscadore, 2018-06-13:
@danielg4 Thanks for answearing. I didn't read the entire README before posting so I deleted the previous post. So now I have the vpn interface with the IP assigned by the gateway although I can't reach any resource at the gateway end. I have 3 zones, lan, wan and vpn and also two forwarding rule lan->vpn and vpn->wan (ISP connected). I think this would suffice but it's not working, do I need to configure something via iptables (I'm avoiding it) what am I doing wrong?

danielg4, 2018-06-15:
It may be more difficult to get the firewall zoning right with more zones. However, if only trusted resources are reachable on the other end of the VPN, you could try just putting the VPN in the LAN zone. Similarly, if only mostly untrusted resources are on the other end, it could make sense to use the WAN zone.

hscadore, 2018-06-15:
Yep, I will keep the two zones scheme for now. It works for me if I just include the vpnc interface to the wan zone. Whats is the point of having option interface 'wan' set? At first I thought it was mandatory because it meant that the tunnel would be estabilished using the wan interface but I just removed this option and it has no influece at all. Am I missing something?

danielg4, 2018-06-15:
It is possible to have more than one WAN interface, e.g. VDSL+LTE. The option allows choosing one for the VPN endpoint.

hscadore, 2018-07-26:
From te README:
"The additional file(s) are also used:
/etc/vpnc/ca-vpn-MYVPN.pem: The server's CA certificate (for auth_mode 'hybrid')"

Does vpnc support client authentication using certificates issued by the server? The way I see it It's only possible to authenticate the server using certificates. Am I right?

dibdot, 2018-08-17:
@danielg4 @hscadore last post of the ticket owner is from 06/2015 ... it might be better to discuss certain aspects of vpnc in the forum - close this endless story now.

closed 2018-08-17 by dibdot