openwrt/packages #1358: ddns-scripts LUCI no-ip multiple hosts shows incorrectly as error

cor-of-org, 2015-06-09:
The correct format for sending multiple hosts to no-ip.com is to use a comma between them, e.g..

foo.no-ip.com,bar,no-ip.com,foobar.no-ip.com

However, if you do this in LUCI, the input turns red to warn you of an error and will not save. Editing /etc.config/ddns directly is the only way to insert the correct entries, which then work fine.

;o) Cor

chris5560, 2015-06-09:
ddns-scripts package is not designed to update multiple hosts in one call, so LuCI-app also did not support it.
If you put a list of hosts into option domain inside /etc/config/ddns the script will fail because the nslookup command could not handle multiple hosts. Please validate on console or inside ddns-scripts logfile.
The correct way to update multiple hosts in one call is to use "option update_url" so a "custom" service provider. Set one of your hosts into "option domain" and hardcode all hosts inside update_url.
To support Multi-Host updates might be an option for the next release of ddns-scripts, but from my current point of view most users are only updating one host.
I personal update 3 hosts using 3 separate sections inside /etc/config/ddns without any problems.

Christian

cor-of-org, 2015-07-04:
I'm not having any issues - when you enter multiple hosts in /etc/config/ddns /it works fine/.

I'm letting folk know that if they are frustrated by LUCI's handling of this, editing the file directly will get you the result you want, using the format no-ip.com expects.

If LUCI /allowed/ the correct multiple hosts entry, that would be even better!

;o) Cor

chris5560, 2015-07-05:
Hi Cor,

I'm very sorry to tell you that if you set "option domain" in your configuration not to a FQDN you will break ddns-scripts, because the entry in "option domain" is used inside "nslookup $domain".
nslookup does not support a comma separated host list to verify. It also make no sense.

It might looks like working because you will send every "option retry_interval" or "option check_interval" an update to no-ip.com. If you do the same with other providers your account will be looked out.
Please verify with your logfile.

The only solution is to use "option update_url" (custom update url) as described above.

Christian

chris5560, 2015-07-21:
@cor-of-org
Hi Cor,
could you please verify the updated description at OpenWrt wiki: http://wiki.openwrt.org/doc/howto/ddns.client#noipcom
Hopefully it's correct and understandable.
Thanks
Christian

cor-of-org, 2015-07-23:
If I run the update command from the shell, I get (anonymized):

update_url=http://[USERNAME]:[PASSWORD]@dynupdate.no-ip.com/nic/update?hostname=[DOMAIN]&myip=[IP]
force seconds = 237600
check seconds = 840
old process id (if it exists) = " 1088 root 1424 S {dynamic_dns_upd} /bin/sh /usr/lib/ddns/dynamic_dns_updater.sh noip 0"
time_since_update = 0 hours
Running IP check...
current system ip = 78.*.*.*
registered domain ip =
update necessary, performing update ...
updating with url="http://me%40gmx.net:PASSWORD@dynupdate.no-ip.com/nic/update?hostname=mydomain.no-ip.com,mydomain.no-ip.biz,mydomain.no-ip.org,mydomain.no-ip.info,mydomain.serveftp.com&myip=78.*.*.*"
Connecting to dynupdate.no-ip.com (8.23.224.120:80)
-                    100% |***************************************************************************************|    99   0:00:00 ETA
Update Output:
nochg 78.*.*.* nochg 78.*.*.* nochg 78.*.*.* nochg 78.*.*.* nochg 78.*.*.*

update complete, time is: Thu Jul 23 11:02:16 BST 2015

It's working fine as-is. There is nothing in my logs to indicate otherwise. In fact, there is no mention of ddns in my logs at all.

The new instructions look fine though I haven't tested them. The point of this post was to ask if multiple hosts functionality could be added to LUCI, not that I use it, but many do.

As far as I know, there isn't even an "update URL" option in LUCI.

;o) Cor

chris5560, 2015-07-24:
Hi Cor,

thanks for feedback. Everything looks ok.
I also can leave the wiki as is.
Inside LuCI-app-ddns, if you select --custom-- service, then the field update_url is shown (since 10.03).

Christian

closed 2016-08-26 by hnyman