openwrt/packages #1359: c-ares (libcares) not buid (error) can use protection to stack buffer overflow

build000, 2015-06-10:
[04:20:46] slawek@slawek-ubuntu:/projekty/openwrt/trunk$ make -j1 package/c-ares/compile V=s
make[1]: Entering directory '/projekty/openwrt/trunk'
make[2]: Entering directory '/projekty/openwrt/trunk/package/libs/toolchain'
if [ -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "libc" >> /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install
if [ -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "libgcc" >> /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install
if [ -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "libssp" >> /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install
if [ -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "libpthread" >> /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install
if [ -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "librt" >> /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install
make[2]: Leaving directory '/projekty/openwrt/trunk/package/libs/toolchain'
make[2]: Entering directory '/projekty/openwrt/trunk/feeds/packages/libs/c-ares'
rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.c-ares_installed
(cd /projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/c-ares-1.10.0; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf autom4te.cache; [ -e ./config.rpath ] || ln -s /projekty/openwrt/trunk/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/projekty/openwrt/trunk/staging_dir/host/bin/autom4te AUTOCONF=/projekty/openwrt/trunk/staging_dir/host/bin/autoconf AUTOMAKE=/projekty/openwrt/trunk/staging_dir/host/bin/automake ACLOCAL=/projekty/openwrt/trunk/staging_dir/host/bin/aclocal AUTOHEADER=/projekty/openwrt/trunk/staging_dir/host/bin/autoheader LIBTOOLIZE=/projekty/openwrt/trunk/staging_dir/host/bin/libtoolize LIBTOOL=/projekty/openwrt/trunk/staging_dir/host/bin/libtool M4=/projekty/openwrt/trunk/staging_dir/host/bin/m4 AUTOPOINT=true /projekty/openwrt/trunk/staging_dir/host/bin/autoreconf -v -f -i -s -B /projekty/openwrt/trunk/staging_dir/host/share/aclocal -I /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/host/share/aclocal -I /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/share/aclocal -I m4 -I . . || true; fi; );
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: /projekty/openwrt/trunk/staging_dir/host/bin/aclocal -I /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/host/share/aclocal -I /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/share/aclocal -I m4 -I . --force -I m4
autoreconf: configure.ac: tracing
autoreconf: running: /projekty/openwrt/trunk/staging_dir/host/bin/libtoolize --force
OpenWrt-libtoolize: putting auxiliary files in `.'.
OpenWrt-libtoolize: linking file `./ltmain.sh'
OpenWrt-libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
OpenWrt-libtoolize: linking file `m4/libtool.m4'
OpenWrt-libtoolize: linking file `m4/ltoptions.m4'
OpenWrt-libtoolize: linking file `m4/ltsugar.m4'
OpenWrt-libtoolize: linking file `m4/ltversion.m4'
OpenWrt-libtoolize: linking file `m4/lt~obsolete.m4'
OpenWrt-libtoolize: Remember to add `LT_INIT' to configure.ac.
autoreconf: running: /projekty/openwrt/trunk/staging_dir/host/bin/autoconf --include=/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/host/share/aclocal --include=/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/share/aclocal --include=m4 --include=. --prepend-include=/projekty/openwrt/trunk/staging_dir/host/share/aclocal --force
autoreconf: running: /projekty/openwrt/trunk/staging_dir/host/bin/autoheader --include=/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/host/share/aclocal --include=/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/share/aclocal --include=m4 --include=. --prepend-include=/projekty/openwrt/trunk/staging_dir/host/share/aclocal --force
autoreconf: running: /projekty/openwrt/trunk/staging_dir/host/bin/automake --add-missing --force-missing
autoreconf: Leaving directory `.'
(cd /projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/c-ares-1.10.0/./; if [ -x ./configure ]; then find /projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/c-ares-1.10.0/ -name config.guess | xargs -r chmod u+w; find /projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/c-ares-1.10.0/ -name config.guess | xargs -r -n1 cp --remove-destination /projekty/openwrt/trunk/scripts/config.guess; find /projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/c-ares-1.10.0/ -name config.sub | xargs -r chmod u+w; find /projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/c-ares-1.10.0/ -name config.sub | xargs -r -n1 cp --remove-destination /projekty/openwrt/trunk/scripts/config.sub; AR="mips-openwrt-linux-uclibc-gcc-ar" AS="mips-openwrt-linux-uclibc-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro" LD=mips-openwrt-linux-uclibc-ld NM="mips-openwrt-linux-uclibc-gcc-nm" CC="mips-openwrt-linux-uclibc-gcc" GCC="mips-openwrt-linux-uclibc-gcc" CXX="mips-openwrt-linux-uclibc-g++" RANLIB="mips-openwrt-linux-uclibc-gcc-ranlib" STRIP=mips-openwrt-linux-uclibc-strip OBJCOPY=mips-openwrt-linux-uclibc-objcopy OBJDUMP=mips-openwrt-linux-uclibc-objdump SIZE=mips-openwrt-linux-uclibc-size CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/projekty/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/projekty/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include " LDFLAGS="-L/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib -L/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/lib -L/projekty/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/projekty/openwrt/trunk/staging_di [... cut ...]
make[2]: *** [/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/c-ares-1.10.0/.configured_yyyyyyyy] Error 1
make[2]: Leaving directory '/projekty/openwrt/trunk/feeds/packages/libs/c-ares'
package/Makefile:187: recipe for target 'package/feeds/packages/c-ares/compile' failed
make[1]: *** [package/feeds/packages/c-ares/compile] Error 2
make[1]: Leaving directory '/projekty/openwrt/trunk'
/projekty/openwrt/trunk/include/toplevel.mk:181: polecenia dla obiektu 'package/c-ares/compile' nie powiodły się
make: *** [package/c-ares/compile] Błąd 2
[04:22:14] slawek@slawek-ubuntu:/projekty/openwrt/trunk$

karlp, 2015-06-16:
Did you have conservative or aggressive for this?

karlp, 2015-06-16:
You can try this patch: https://github.com/remakeelectric/packages/commit/cdde48d76cf1ba5519b793ba890c284636385a74

However, I'm not sure it's the right way.  Perhaps the fortify source should be setting CPPFLAGS?

build000, 2015-06-16:
@karip - jusr aggressive - like this: https://github.com/openwrt/packages/issues/1321

karlp, 2015-07-10:
this is fixed via jow's patch

closed 2015-07-10 by karlp