openwrt/packages #1366: Adding python3 modules

currywurst76, 2015-06-14:
How to add python3 modules/packages for the following components:

* python3-curl
* python3-crypto
* python3-expat
* pytohn3-imaging-library

commodo, 2015-06-14:
Well, you'd need to create a Makefile for each, and put them in the `lang` folder.
Please check this : https://github.com/openwrt/packages/blob/master/CONTRIBUTING.md

You can also check this repo: https://www.gitorious.org/openwrt-cybjit/python3/source/866e87b9912acea24871a68b503ba003cc788da1:lang
Some Python3 packages have already been ported to OpenWRT by cybjit.
If you want to copy them over to this packages repo, a maintainer must be found for each package.

hnyman, 2016-08-23:
no activity for a year

closed 2016-08-23 by hnyman