openwrt/packages #1370: sizeof(off_t) in Erlang

dangowrt, 2015-06-14:
Erlang uses autotools and a precompiler macro to detect sizeof(off_t) which doesn't work when cross-compiling.
This can be fixed by either detecting sizeof(off_t) in run-time like
https://git-us.netdef.org/projects/OSR/repos/openwrt-isis-hnet/browse/erlang/patches/105-use_correct_off_t.patch

or by appending CFLAGS with `-D_FILE_OFFSET_BITS=64`

dangowrt, 2015-06-14:
@psycho-nico

dangowrt, 2015-06-14:
@cfra can give more background if needed

psycho-nico, 2016-08-19:
> Erlang uses autotools and a precompiler macro to detect sizeof(off_t) which doesn't work when cross-compiling.

The configure script is not checking sizeof(off_t), it is just using predefined values from an appropriate site config file in `${OPENWRT}/include/site/` directory.

```
configure: loading site script /fun/Projects/OpenWrt/include/site/arm
...
checking for off_t... (cached) yes
checking size of off_t... (cached) 8
```

I'm closing the issue since I don't see what is wrong with the above.

Cheers,
-{Nico}

closed 2016-08-19 by psycho-nico

dangowrt, 2016-08-19:
Things like reading the size of a file don't work because Erlang wrongly assumes sizeof(off_t) to be 8 bytes, which is not true on ARM, MIPS and probably other 32-bit targets without PAE. You'd either need the before-mentioned patch which detects sizeof(off_t) at runtime or always enable support for very large files using `-D_FILE_OFFSET_BITS=64` to enforce sizeof(off_t) to be 8 bytes.
However, this bug report is rather old and things may have been addressed upstream. I haven't recently tested Erlang myself.

psycho-nico, 2016-08-23:
Humm, you're right, the bug report is old and does not seem applicable anymore since the move to musl.

From the [musl FAQ](http://wiki.musl-libc.org/wiki/FAQ#Q:_do_i_need_to_define_LARGEFILE64_SOURCE_to_get_64bit_off_t_.3F):
> under musl, off_t is always 64-bit.

Cheers,
-{Nico}

gotthardp, 2017-02-13:
I'm afraid the problem persists. If "under musl, off_t is always 64-bit", then the problem persists, because as stated above this is "not true on ARM, MIPS and probably other 32-bit targets without PAE".