openwrt/packages #1372: Updated vpnc-scripts package breaks vpn setups with multiple dns servers and split dns

sykesm, 2015-06-15:
The implementation of `vpnc-script` is supposed to get 0..n DNS server addresses from `INTERNAL_IP4_DNS` and `INTERNAL_IP6_DNS` but the implementation that moved from openconnect to vpnc-scripts only supports one in a split DNS configuration.

In particular, the code here: https://github.com/openwrt/packages/blob/d16359b96e8e574925af1af42e08777624aceb3a/net/vpnc-scripts/files/vpnc-script#L84-L97 assumes `INTERNAL_IP4_DNS` and `INTERNAL_IP6_DNS` only contain one IP; when multiple servers are present, it generates a broken configuration that requires manual intervention to correct.

It's also unfortunate that the DNS server is bounced when the configuration changes when use of the `servers-file` option in dnsmasq would avoid that.

marcosscriven, 2015-06-19:
I'm also interested in this issue - our VPN has two DNS servers, and I find resolving internal domains fail.

If I simply stop dnsmasq, the contents of ```/tmp/resolv.conf.auto``` get copied to ```/etc/resolv.conf```, and VPN DNS resolution works fine. Unfortunately as @sykesm says, dnsmasq is bounced, so if I want to use this as a solution I have to stop dnsmasq every time I bring up the VPN.

nmav, 2015-06-24:
I understand the issue with multiple servers, but not what is meant with dnsmasq is bounced.

PR #1489 (2015-06-27): vpnc-script: handle multiple servers in INTERNAL_IP_DNS vars
Resolves #1372
commit 39ecf00d (2015-06-27): vpnc-script: handle multiple servers in INTERNAL_IP_DNS vars

Resolves #1372
commit 814f65cf (2015-06-27): vpnc-script: handle multiple servers in INTERNAL_IP_DNS vars

Resolves #1372
closed 2015-06-27 by nmav