openwrt/packages #1375: sstp-client build error

taeasy, 2015-06-15:
make[5]: Entering directory `/home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/sstp-client-1.0.10/src'
mips-openwrt-linux-uclibc-gcc -DHAVE_CONFIG_H -I. -I..   -I/home/main/ar71xx/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/home/main/ar71xx/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/home/main/ar71xx/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/main/ar71xx/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include  -I../include -DSSTP_RUNTIME_DIR='"/var/run/sstpc"' -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16   -I/home/main/ar71xx/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include  -MT sstpc-sstp-client.o -MD -MP -MF .deps/sstpc-sstp-client.Tpo -c -o sstpc-sstp-client.o `test -f 'sstp-client.c' || echo './'`sstp-client.c
mv -f .deps/sstpc-sstp-client.Tpo .deps/sstpc-sstp-client.Po
mips-openwrt-linux-uclibc-gcc -DHAVE_CONFIG_H -I. -I..   -I/home/main/ar71xx/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/home/main/ar71xx/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/home/main/ar71xx/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/main/ar71xx/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include  -I../include -DSSTP_RUNTIME_DIR='"/var/run/sstpc"' -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16   -I/home/main/ar71xx/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include  -MT sstpc-sstp-option.o -MD -MP -MF .deps/sstpc-sstp-option.Tpo -c -o sstpc-sstp-option.o `test -f 'sstp-option.c' || echo './'`sstp-option.c
In file included from sstp-private.h:69:0,
                 from sstp-option.c:33:
sstp-chap.h:65:9: error: unknown type name 'uint8_t'
         uint8_t skey[16], uint8_t rkey[16], char server);
         ^
sstp-chap.h:65:27: error: unknown type name 'uint8_t'
         uint8_t skey[16], uint8_t rkey[16], char server);
                           ^
In file included from sstp-private.h:70:0,
                 from sstp-option.c:33:
sstp-state.h:64:54: error: unknown type name 'uint8_t'
 typedef status_t (*sstp_state_forward_fn)(void *arg, uint8_t *data, 
                                                      ^
sstp-state.h:89:51: error: unknown type name 'sstp_state_forward_fn'
 void sstp_state_set_forward(sstp_state_st *state, sstp_state_forward_fn 
                                                   ^
In file included from sstp-private.h:73:0,
                 from sstp-option.c:33:
sstp-event.h:54:53: error: unknown type name 'uint8_t'
 status_t sstp_event_mppe_result(sstp_event_st *ctx, uint8_t **skey, 
                                                     ^
sstp-event.h:55:23: error: unknown type name 'uint8_t'
         size_t *slen, uint8_t **rkey, size_t *rlen);
                       ^
In file included from sstp-private.h:75:0,
                 from sstp-option.c:33:
sstp-cmac.h:45:5: error: unknown type name 'uint8_t'
     uint8_t key[32];
     ^
sstp-cmac.h:48:5: error: unknown type name 'uint8_t'
     uint8_t seed[32];
     ^
sstp-cmac.h:51:5: error: unknown type name 'uint8_t'
     uint8_t slen;
     ^
sstp-cmac.h:68:43: error: unknown type name 'uint8_t'
 void sstp_cmac_send_key(cmac_ctx_st *ctx, uint8_t *key, int len);
                                           ^
sstp-cmac.h:74:43: error: unknown type name 'uint8_t'
 void sstp_cmac_recv_key(cmac_ctx_st *ctx, uint8_t *key, int len);
                                           ^
sstp-cmac.h:84:41: error: unknown type name 'uint8_t'
 void sstp_cmac_result(cmac_ctx_st *ctx, uint8_t *msg, int mlen, uint8_t *result, int length);
                                         ^
sstp-cmac.h:84:65: error: unknown type name 'uint8_t'
 void sstp_cmac_result(cmac_ctx_st *ctx, uint8_t *msg, int mlen, uint8_t *result, int length);
                                                                 ^
In file included from sstp-private.h:76:0,
                 from sstp-option.c:33:
sstp-packet.h:155:5: error: unknown type name 'uint8_t'
     uint8_t version;
     ^
sstp-packet.h:158:5: error: unknown type name 'uint8_t'
     uint8_t flags;
     ^
sstp-packet.h:161:5: error: unknown type name 'uint16_t'
     uint16_t length;
     ^
sstp-packet.h:175:5: error: unknown type name 'uint8_t'
     uint8_t reserved;
     ^
sstp-packet.h:178:5: error: unknown type name 'uint8_t'
     uint8_t type;
     ^
sstp-packet.h:181:5: error: unknown type name 'uint16_t'
     uint16_t length;
     ^
sstp-packet.h:195:5: error: unknown type name 'uint16_t'
     uint16_t type;
     ^
sstp-packet.h:198:5: error: unknown type name 'uint16_t'
     uint16_t nattr;
     ^
sstp-packet.h:222:1: error: unknown type name 'uint8_t'
 uint8_t *sstp_pkt_data(sstp_buff_st *buf);
 ^
make[5]: *** [sstpc-sstp-option.o] Error 1
make[5]: Leaving directory `/home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/sstp-client-1.0.10/src'
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory `/home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/sstp-client-1.0.10'
make[3]: *** [all] Error 2
make[3]: Leaving directory `/home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/sstp-client-1.0.10'
make[2]: *** [/home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/sstp-client-1.0.10/.built] Error 2
make[2]: Leaving directory `/home/main/ar71xx/feeds/packages/net/sstp-client'
make[1]: *** [package/feeds/packages/sstp-client/compile] Error 2
make[1]: Leaving directory `/home/main/ar71xx'
make: *** [package/sstp-client/compile] Error 2
main@bogon:~/ar71xx$

cindywijaya, 2015-08-06:
try to add
include <stdint.h>
on top of
sstp-chap.h
sstp-option.c

alzhao, 2015-11-07:
patch 000-fix-unit8_t.patch. How can we add this to github?

--- a/src/sstp-chap.h
+++ b/src/sstp-chap.h
@@ -24,7 +24,7 @@
 
 #ifndef __SSTP_CHAP_H__
 #define __SSTP_CHAP_H__
-
+#include <stdint.h>
 
 #define SSTP_CHAP_SENDING   0x01
 #define SSTP_CHAP_SERVER    0x02
--- a/src/sstp-cmac.h
+++ b/src/sstp-cmac.h
@@ -25,7 +25,7 @@
 #ifndef __SSTP_CMAC_H__
 #define __SSTP_CMAC_H__
 
-
+#include <stdint.h>
 /*< Set SHA1 operation for the crypto binding */
 #define SSTP_CMAC_SHA1      SSTP_PROTO_HASH_SHA1
 
--- a/src/sstp-event.h
+++ b/src/sstp-event.h
@@ -24,6 +24,7 @@
 #ifndef __SSTP_EVENT_H__
 #define __SSTP_EVENT_H__
 
+#include <stdint.h>
 /*< Forward declare event context */
 struct sstp_event;
 typedef struct sstp_event sstp_event_st;
--- a/src/sstp-packet.h
+++ b/src/sstp-packet.h
@@ -25,7 +25,7 @@
 #ifndef __SSTP_PACKET_H__
 #define __SSTP_PACKET_H__
 
-
+#include <stdint.h>
 /*< The protocol encapsulation, always PPP */
 #define SSTP_ENCAP_PROTO_PPP    0x0001
 
--- a/src/sstp-state.h
+++ b/src/sstp-state.h
@@ -24,6 +24,7 @@
 #ifndef __SSTP_STATE_H__
 #define __SSTP_STATE_H__
 
+#include <stdint.h>
 
 #define SSTP_ST_CALL_CONNECT_REQ       0x0004
 #define SSTP_ST_CALL_CONNECT_ACK       0x0008

hnyman, 2016-08-26:
sstp-client builds surrently ok trunk buildbot

closed 2016-08-26 by hnyman