openwrt/packages #1376: libdaemon fails to build with musl

teknoraver, 2015-06-15:
libdaemon fails to build with musl as C library:
```
make[5]: Entering directory '/home/matteo/src/openwrt/build_dir/target-mips_34kc+dsp_musl-1.1.9/libdaemon-0.14/examples'
  CC       testd.o
testd.c:26:24: fatal error: sys/unistd.h: No such file or directory
 #include <sys/unistd.h>
                        ^
compilation terminated.
```

dangowrt, 2015-06-15:
You got to replace
#include <sys/unistd.h>
by
#include <unistd.h>

That's supposed to work with all libc implementations.

thess, 2015-06-15:
Problem was fixed in 15.05 & trunk. Remove building of examples by adding:
>--disable-examples

to configure args in Makeflle. See c04f69d52785138e67ca0e1bf469368b1fe4b7f3

closed 2015-06-18 by thess