openwrt/packages #1381: etherwake fails to compile against musl

joaochainho, 2015-06-16:
Hi,

etherwake fails to compile against musl.
Arch: ar71xx
Rev: trunk r45995

```
make[1]: Entering directory `/home/openwrt/trunk'
make[2]: Entering directory `/home/openwrt/trunk/package/libs/toolchain'
if [ -f /home/openwrt/trunk/staging_dir/target-mips_24kc_musl-1.1.9/pkginfo/toolchain.default.install.clean ]; then rm -f /home/openwrt/trunk/staging_dir/target-mips_24kc_musl-1.1.9/pkginfo/toolchain.default.install /home/openwrt/trunk/staging_dir/target-mips_24kc_musl-1.1.9/pkginfo/toolchain.default.install.clean; fi; echo "libc" >> /home/openwrt/trunk/staging_dir/target-mips_24kc_musl-1.1.9/pkginfo/toolchain.default.install
if [ -f /home/openwrt/trunk/staging_dir/target-mips_24kc_musl-1.1.9/pkginfo/toolchain.default.install.clean ]; then rm -f /home/openwrt/trunk/staging_dir/target-mips_24kc_musl-1.1.9/pkginfo/toolchain.default.install /home/openwrt/trunk/staging_dir/target-mips_24kc_musl-1.1.9/pkginfo/toolchain.default.install.clean; fi; echo "libgcc" >> /home/openwrt/trunk/staging_dir/target-mips_24kc_musl-1.1.9/pkginfo/toolchain.default.install
if [ -f /home/openwrt/trunk/staging_dir/target-mips_24kc_musl-1.1.9/pkginfo/toolchain.default.install.clean ]; then rm -f /home/openwrt/trunk/staging_dir/target-mips_24kc_musl-1.1.9/pkginfo/toolchain.default.install /home/openwrt/trunk/staging_dir/target-mips_24kc_musl-1.1.9/pkginfo/toolchain.default.install.clean; fi; echo "libpthread" >> /home/openwrt/trunk/staging_dir/target-mips_24kc_musl-1.1.9/pkginfo/toolchain.default.install
if [ -f /home/openwrt/trunk/staging_dir/target-mips_24kc_musl-1.1.9/pkginfo/toolchain.default.install.clean ]; then rm -f /home/openwrt/trunk/staging_dir/target-mips_24kc_musl-1.1.9/pkginfo/toolchain.default.install /home/openwrt/trunk/staging_dir/target-mips_24kc_musl-1.1.9/pkginfo/toolchain.default.install.clean; fi; echo "librt" >> /home/openwrt/trunk/staging_dir/target-mips_24kc_musl-1.1.9/pkginfo/toolchain.default.install
make[2]: Leaving directory `/home/openwrt/trunk/package/libs/toolchain'
make[2]: Entering directory `/home/openwrt/trunk/feeds/packages/net/etherwake'
mips-openwrt-linux-musl-gcc -Os -pipe -march=24kc -mtune=24kc -mno-branch-likely -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -D__UCLIBC__ /home/openwrt/trunk/build_dir/target-mips_24kc_musl-1.1.9/etherwake-1.09.orig/ether-wake.c -o /home/openwrt/trunk/build_dir/target-mips_24kc_musl-1.1.9/etherwake-1.09.orig/etherwake
In file included from /home/openwrt/trunk/staging_dir/toolchain-mips_24kc_gcc-4.9-linaro_musl-1.1.9/include/netinet/ether.h:8:0,
                 from /home/openwrt/trunk/build_dir/target-mips_24kc_musl-1.1.9/etherwake-1.09.orig/ether-wake.c:94:
/home/openwrt/trunk/staging_dir/toolchain-mips_24kc_gcc-4.9-linaro_musl-1.1.9/include/netinet/if_ether.h:96:8: error: redefinition of 'struct ethhdr'
 struct ethhdr {
        ^
In file included from /home/openwrt/trunk/build_dir/target-mips_24kc_musl-1.1.9/etherwake-1.09.orig/ether-wake.c:91:0:
/home/openwrt/trunk/staging_dir/toolchain-mips_24kc_gcc-4.9-linaro_musl-1.1.9/include/linux/if_ether.h:137:8: note: originally defined here
 struct ethhdr {
        ^
make[2]: *** [/home/openwrt/trunk/build_dir/target-mips_24kc_musl-1.1.9/etherwake-1.09.orig/.built] Error 1
make[2]: Leaving directory `/home/openwrt/trunk/feeds/packages/net/etherwake'
make[1]: *** [package/feeds/packages/etherwake/compile] Error 2
make[1]: Leaving directory `/home/openwrt/trunk'
make: *** [package/etherwake/compile] Error 2
```

hnyman, 2015-06-16:
Same here, with up-to-date trunk r46001 with musl 1.1.10+git.
ar71xx  & standard gcc 4.8 : 
```
In file included from /Openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/include/netinet/ether.h:8:0,
                 from /Openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/etherwake-1.09.orig/ether-wake.c:94:
/Openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/include/netinet/if_ether.h:96:8: error: redefinition of 'struct ethhdr'
 struct ethhdr {
        ^
In file included from /Openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/etherwake-1.09.orig/ether-wake.c:91:0:
/Openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/include/linux/if_ether.h:137:8: note: originally defined here
 struct ethhdr {
        ^
Makefile:55: recipe for target '/Openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/etherwake-1.09.orig/.built' failed

```

jow-, 2015-06-16:
See #1387

joaochainho, 2015-06-16:
Thanks for quickly solving this issue. It's running fine on the real device.

closed 2015-06-23 by Noltari