openwrt/packages #1382: libupnp fail to compile in parallel, works in rebuild

karlp, 2015-06-16:
Working on the BB/14.07 branch, but the version is the same.  make clean; make -j8 V=S gets this failure _sometimes_  (Sometimes it's another library, with the same autotools error)  make package/libupnp/{clean,compile} will succeed however.   I'd like to know what needs to be specified better to make it build properly parallel

Full log   http://paste.fedoraproject.org/232501/14344495/

snippet from the tail
```
configure: creating ./config.status
config.status: creating Makefile
config.status: creating ixml/Makefile
config.status: creating ixml/doc/Makefile
config.status: creating threadutil/Makefile
config.status: creating upnp/Makefile
config.status: creating upnp/doc/Makefile
config.status: creating upnp/sample/Makefile
config.status: creating docs/dist/Makefile
config.status: creating libupnp.pc
config.status: creating autoconfig.h
config.status: creating upnp/inc/upnpconfig.h
config.status: executing depfiles commands
config.status: executing libtool commands
configure: WARNING: unrecognized options: --disable-nls
configure: copying "autoconfig.h"          to "build/inc/autoconfig.h"
configure: copying "upnp/inc/upnpconfig.h" to "build/inc/upnpconfig.h"
rm -f /home/karlp/src/smartgate_firmware/openwrt-bb/build_dir/target-mips_34kc_uClibc-0.9.33.2/libupnp-1.6.19/.configured_
touch /home/karlp/src/smartgate_firmware/openwrt-bb/build_dir/target-mips_34kc_uClibc-0.9.33.2/libupnp-1.6.19/.configured_
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16  -I/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16  -I/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include " LDFLAGS="-L/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib -L/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/target-mips_34kc_uClibc-0.9.33.2/lib -L/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib " make -j1 -C /home/karlp/src/smartgate_firmware/openwrt-bb/build_dir/target-mips_34kc_uClibc-0.9.33.2/libupnp-1.6.19/. AR=mips-openwrt-linux-uclibc-ar AS="ccache_cc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float" LD=mips-openwrt-linux-uclibc-ld NM=mips-openwrt-linux-uclibc-nm CC="ccache_cc" GCC="ccache_cc" CXX="ccache_cxx" RANLIB=mips-openwrt-linux-uclibc-ranlib STRIP=mips-openwrt-linux-uclibc-strip OBJCOPY=mips-openwrt-linux-uclibc-objcopy OBJDUMP=mips-openwrt-linux-uclibc-objdump SIZE=mips-openwrt-linux-uclibc-size CROSS="mips-openwrt-linux-uclibc-" ARCH="mips" ;
make[4]: Entering directory '/home/karlp/src/smartgate_firmware/openwrt-bb/build_dir/target-mips_34kc_uClibc-0.9.33.2/libupnp-1.6.19'
CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/sh /home/karlp/src/smartgate_firmware/openwrt-bb/build_dir/target-mips_34kc_uClibc-0.9.33.2/libupnp-1.6.19/build-aux/missing --run aclocal-1.12 -I m4
/home/karlp/src/smartgate_firmware/openwrt-bb/build_dir/target-mips_34kc_uClibc-0.9.33.2/libupnp-1.6.19/build-aux/missing: line 51: aclocal-1.12: command not found
WARNING: 'aclocal-1.12' is missing on your system.  You should only need it if
         you modified 'acinclude.m4' or 'configure.ac'.  You might want
         to install the Automake and Perl packages.  Grab them from
         any GNU archive site.
 cd . && /bin/sh /home/karlp/src/smartgate_firmware/openwrt-bb/build_dir/target-mips_34kc_uClibc-0.9.33.2/libupnp-1.6.19/build-aux/missing --run automake-1.12 --foreign
/home/karlp/src/smartgate_firmware/openwrt-bb/build_dir/target-mips_34kc_uClibc-0.9.33.2/libupnp-1.6.19/build-aux/missing: line 51: automake-1.12: command not found
WARNING: 'automake-1.12' is missing on your system.  You should only need it if
         you modified 'Makefile.am', 'acinclude.m4' or 'configure.ac'.
         You might want to install the Automake and Perl packages.
         Grab them from any GNU archive site.
CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/sh /home/karlp/src/smartgate_firmware/openwrt-bb/build_dir/target-mips_34kc_uClibc-0.9.33.2/libupnp-1.6.19/build-aux/missing --run autoconf
configure.ac:362: error: possibly undefined macro: AM_INIT_AUTOMAKE
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:429: error: possibly undefined macro: AM_CONDITIONAL
configure.ac:523: error: possibly undefined macro: AM_PROG_CC_C_O
configure.ac:525: error: possibly undefined macro: AC_PROG_LIBTOOL
configure.ac:609: error: possibly undefined macro: AC_MSG_ERROR
Makefile:382: recipe for target 'configure' failed
make[4]: *** [configure] Error 1
make[4]: Leaving directory '/home/karlp/src/smartgate_firmware/openwrt-bb/build_dir/target-mips_34kc_uClibc-0.9.33.2/libupnp-1.6.19'
Makefile:75: recipe for target '/home/karlp/src/smartgate_firmware/openwrt-bb/build_dir/target-mips_34kc_uClibc-0.9.33.2/libupnp-1.6.19/.built' failed
make[3]: *** [/home/karlp/src/smartgate_firmware/openwrt-bb/build_dir/target-mips_34kc_uClibc-0.9.33.2/libupnp-1.6.19/.built] Error 2
make[3]: Leaving directory '/home/karlp/src/smartgate_firmware/openwrt-bb/feeds/packages/libs/libupnp'
```

karlp, 2015-06-16:
@thess This is your package, any ideas?

thess, 2015-06-16:
Does setting _PKG_BUILD_PARALLEL:=0_ help? It's a pretty straightforward package build.
I'll need to compare build log success with failure for any hints.

What other packages show this symptom?

karlp, 2015-06-16:
I've seen it a few times with libcares / c-ares too.  I can't reproduce it on demand, so I can't tell whether PKG_BUILD_PARALLEL has actually helped or not. :(

diizzyy, 2017-06-07:
@karlp 
Still broken?

diizzyy, 2018-06-11:
@karlp 
Ping?

dibdot, 2018-08-18:
seems to be fixed.

closed 2018-08-18 by dibdot