openwrt/packages #1383: picocom fails to compile with musl

joaochainho, 2015-06-16:
Hi,
picocom fails to compile with musl as C library.
Arch: ar71xx
Rev: trunk r45995

```
make[3]: Entering directory `/home/openwrt/trunk/build_dir/target-mips_24kc_musl-1.1.9/picocom-1.7'
mips-openwrt-linux-musl-gcc -Os -pipe -march=24kc -mtune=24kc -mno-branch-likely -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -DVERSION_STR=\"1.7\" -DUUCP_LOCK_DIR=\"/var/lock\" -DHIGH_BAUD  -c -o term.o term.c
term.c:37:20: fatal error: termio.h: No such file or directory
 #include <termio.h>
                    ^
compilation terminated.
make[3]: *** [term.o] Error 1
```

closed 2015-06-16 by jow-

joaochainho, 2015-06-16:
Thanks for quickly solving this issue. It's running fine on the real device.

karlp, 2015-06-16:
fwiw, filed upstream as https://code.google.com/p/picocom/issues/detail?id=28