openwrt/packages #1384: socat fails to compile with musl

joaochainho, 2015-06-16:
Hi,
socat fails to compile with musl as C library.
Arch: ar71xx
Rev: trunk r45995

```
make[3]: Entering directory `/home/openwrt/trunk/build_dir/target-mips_24kc_musl-1.1.9/socat-1.7.3.0'
mips-openwrt-linux-musl-gcc -Os -pipe -march=24kc -mtune=24kc -mno-branch-likely -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16  -D_GNU_SOURCE -Wall -Wno-parentheses  -DHAVE_CONFIG_H -I. -I/home/openwrt/trunk/staging_dir/target-mips_24kc_musl-1.1.9/usr/include -I/home/openwrt/trunk/staging_dir/target-mips_24kc_musl-1.1.9/include -I/home/openwrt/trunk/staging_dir/toolchain-mips_24kc_gcc-4.9-linaro_musl-1.1.9/usr/include -I/home/openwrt/trunk/staging_dir/toolchain-mips_24kc_gcc-4.9-linaro_musl-1.1.9/include  -I. -I/home/openwrt/trunk/staging_dir/target-mips_24kc_musl-1.1.9/usr/include -I/home/openwrt/trunk/staging_dir/target-mips_24kc_musl-1.1.9/include -I/home/openwrt/trunk/staging_dir/toolchain-mips_24kc_gcc-4.9-linaro_musl-1.1.9/usr/include -I/home/openwrt/trunk/staging_dir/toolchain-mips_24kc_gcc-4.9-linaro_musl-1.1.9/include   -c -o socat.o socat.c
In file included from /home/openwrt/trunk/staging_dir/toolchain-mips_24kc_gcc-4.9-linaro_musl-1.1.9/include/linux/if_tun.h:20:0,
                 from sysincludes.h:145,
                 from socat.c:11:
/home/openwrt/trunk/staging_dir/toolchain-mips_24kc_gcc-4.9-linaro_musl-1.1.9/include/linux/if_ether.h:137:8: error: redefinition of 'struct ethhdr'
 struct ethhdr {
        ^
In file included from sysincludes.h:142:0,
                 from socat.c:11:
/home/openwrt/trunk/staging_dir/toolchain-mips_24kc_gcc-4.9-linaro_musl-1.1.9/include/netinet/if_ether.h:96:8: note: originally defined here
 struct ethhdr {
        ^
make[3]: *** [socat.o] Error 1
```

thess, 2015-06-16:
Closed by #1386

closed 2015-06-16 by thess

joaochainho, 2015-06-16:
Thanks for quickly solving this issue. It's running fine on the real device.