openwrt/packages #1394: FreeRadius @ Chaos Calmer RC2 - Multiple problems

angelosv, 2015-06-16:
**First issue:**
I added a user in /etc/freeradius2/users but I get this error in radiusd.log:

`Mon Jun 15 15:49:47 2015 : Auth: Invalid user: [angelos] (from client localhost port 1 cli XX-XX-XX-XX-XX-XX)`

**Second issue:**
When I run "radiusd -X" I get this error:

`Mon Jun 15 15:41:26 2015 : Error: Failed binding to authentication interface br-lan address * port 1812: Address already in use`
`Mon Jun 15 15:41:26 2015 : Error: /etc/freeradius2/radiusd.conf[273]: Error binding to port for 0.0.0.0 port 1812`

I run the netstat command to see what app is using port 1812, and I get this:

`udp        0      0 192.168.1.1:1812        0.0.0.0:*                           3932/radiusd`
`udp        0      0 192.168.1.1:37588       192.168.1.1:1812        ESTABLISHED 3726/hostapd`

I use exactly the same configuration I used in Barrier Breaker (it worked perfectly in BB).
The router is a WBMR-HP-G300H.

dangowrt, 2015-06-17:
Regarding the first issue:
please check if the configuration files in `/etc/freeradius2` are indentical to what you were using on bb.

Seconds issue:
radiusd is started using an init script, the error you see simply means that it's already running which is also what netstat shows.

angelosv, 2015-06-17:
So I guess the second issue is not a problem, right?

As far as for the first issue, I used **Total Commander** in Windows in order to compare the files both from Barrier Breaker and Chaos Calmer RC2 that are involved in my FreeRadius config...

```
/etc/freeradius2/clients.conf
/etc/freeradius2/eap.conf
/etc/freeradius2/radiusd.conf
/etc/freeradius2/users
/etc/init.d/radiusd
```
All files are identical.

I follow the same steps and do exactly the same changes in those files as I did in BB, but authentication fails.

Please let me know if I can do something to help track this issue.

czakey, 2015-09-12:
Please post your config files. You can cut passwords of course. My radius is working fine.

yousong, 2016-08-19:
Any news on this one?

diizzyy, 2017-06-07:
Please see #1193

yousong, 2018-01-19:
Close as freeradius2 is now abandoned state by 5455a88d45397bf9bdb4c87d087e93ab57709619

closed 2018-01-19 by yousong