openwrt/packages #1398: bwm-ng fails to compile with gcc format-security enabled

joaochainho, 2015-06-17:
Since trunk r46003, bwm-ng fails to compile with gcc format-security enabled.

```
output.c: In function 'print_header':
output.c:237:13: error: format not a string literal and no format arguments [-Werror=format-security]
             printf(input2str());
             ^
cc1: some warnings being treated as errors
make[5]: *** [output.o] Error 1
```

I would like to suggest a patch that solves this issue.

```diff
--- a/src/output.c
+++ b/src/output.c
@@ -234,7 +234,7 @@ int print_header(int option) {
 			if (output_method==PLAIN_OUT && ansi_output) printf("\033[1;2H");
 	        printf("bwm-ng v" VERSION " (delay %2.3fs); ",(float)delay/1000);
 			if (output_method==PLAIN_OUT) printf("press 'ctrl-c' to end this%s",(ansi_output ? "\033[2;2H" : "")); else printf("input: ");
-            printf(input2str());
+            printf("%s\n",input2str());
             printf("%s\n",show_all_if2str());
 			if (output_method==PLAIN_OUT) {
 				if (ansi_output)
```

hnyman, 2015-06-17:
@Zokormazo

Zokormazo, 2015-06-17:
I've just been father and don't have too much time now. Can someone look it? Otherwise I'll try to make time this weekend

Sorry :S

Zokormazo, 2015-06-18:
I make some time to test.

Thanks @joaochainho for report & patching ;)

https://github.com/openwrt/packages/pull/1421

Zokormazo, 2015-07-27:
This is fixed: https://github.com/openwrt/packages/commit/5aae478e24547d79cfefac0fe3562dc0b93b7f21

closed 2015-07-27 by champtar