openwrt/packages #1409: libpam md5sum hash not valid

NucleaPeon, 2015-06-17:
Using the latest packages version, updated June 17 14:25 PST.

`libpam` expects the hash `ee4a480d77b341c99e8b1375f8f180c0`, but is supplied with hash `35b6091af95981b1b2cd60d813b5e4ee`.

I modified the Makefile to use `ee4a480d77b341c99e8b1375f8f180c0` and it continues to build fine, please verify that the new libpam 1.2.0 hash matches and update if necessary.

Thank you

NucleaPeon, 2015-06-17:
If this is an issue I should report to the maintainer, please let me know. I'd hate to spam you guys.

yousong, 2015-06-18:
@nmav

sorry,  md5sum of the tarball was not updated in the previous version bump of libpam.

nmav, 2015-06-18:
@yousong Have you tried the new pam?

yousong, 2015-06-18:
@nmav 

`libpam` is depended by openssh-client.  So I compiled tested it.  The MD5 checksum issue was not triggered in my local env because the tarball was manually downloaded and placed it into the `dl/` directory...

I just compile tested with x86_64 target and confirmed that issue #1417 is related to the updated `/lib64` change.

Sorry for the inconvenience. :(

nmav, 2015-06-18:
Np. It is just that I'll not be able to fix it the next few days. If you can provide a fix (or a rollback) would be nice.

yousong, 2015-06-18:
@nmav 

I created a pull request, #1420

nmav, 2015-06-20:
This should be resolved now; please reopen if not.

closed 2015-06-20 by nmav