openwrt/packages #1417: libpam-1.2.0 fails to build

enigmagroup, 2015-06-18:
After fixing the [wrong md5sum of libpam](https://github.com/openwrt/packages/issues/1409), I run into the following error:

```
cp: cannot stat '/home/x/openwrt-64/build_dir/target-x86_64_uClibc-0.9.33.2/Linux-PAM-1.2.0/ipkg-install/lib/*.so*': No such file or directory
Makefile:73: recipe for target '/home/x/openwrt-64/bin/x86-uClibc/packages/packages/libpam_1.2.0-1_x86_64.ipk' failed
```

yousong, 2015-06-18:
Hi, @enigmagroup 

Is there a `/home/x/openwrt-64/build_dir/target-x86_64_uClibc-0.9.33.2/Linux-PAM-1.2.0/ipkg-install/lib64/`?  If that is the case, then updated patch `0001-build-use-host_cpu-for-lib64-directory-handling.patch` should be the cause.

nmav, 2015-06-20:
This should be resolved now; please reopen if not.

closed 2015-06-20 by nmav