openwrt/packages #1418: gcc fails to build with musl

vbmithr, 2015-06-18:
```
Configuring in host-mips-openwrt-linux-musl/gmp
configure: loading site script /home/vb/code/openwrt/include/site/mips
configure: loading cache ./config.cache
checking build system type... x86_64-unknown-linux-gnu
checking host system type... Invalid configuration `none-openwrt-linux-musl': machine `none-openwrt-linux' not recognized
configure: error: /bin/sh ../.././gmp/config.sub none-openwrt-linux-musl failed
Makefile:4311: recipe for target 'configure-gmp' failed
make[5]: *** [configure-gmp] Error 1
make[5]: Leaving directory '/home/vb/code/openwrt/build_dir/target-mips_34kc_musl-1.1.10/gcc-4.8.3'
Makefile:863: recipe for target 'all' failed
```

vbmithr, 2015-06-20:
The issue happens because the config.{sub,guess} files found in some subdirs of gcc are not aware of musl. I was able to compile gcc when I replaced those files by the ones found at the root of the gcc dir.

quanpower, 2015-06-27:
Thank you,vbmithr!I have the same problm too,and it worked as you say!

vbmithr, 2015-06-28:
np :)

oskarirauta, 2015-07-12:
Could someone post a patch for this..? Or better instructions..?

thess, 2015-07-12:
ping @bk138

bk138, 2015-07-13:
@vbmithr do you have a patch at hand?

vbmithr, 2015-07-14:
On 13/07/2015 09:46, Christian Beier wrote:
> @vbmithr <https://github.com/vbmithr> do you have a patch at hand?

No, I just modified by hands the files. I don't have the time to work on
this right now but could give it a go in a week or so.

bk138, 2015-07-16:
Seems it's enough to copy config.sub to mpfr, gmp and mpc subdirs.

karlp, 2016-05-24:
fixed in 82e56d1bf871712e9baa5c8199e12bec210cfa8e

closed 2016-05-24 by karlp