openwrt/packages #1423: collectd imprecise traffic

basinilya, 2015-06-18:
was here: https://dev.openwrt.org/ticket/19851

Collectd shows greater traffic values than ifconfig

After a few hours it can show 10gb transfered, although the actual traffic was 2gb.

[image]

    root@OpenWrt:~# uptime

        21:43:59 up 2 days, 1:52, load average: 0.16, 0.10, 0.06

    root@OpenWrt:~# ifconfig wwan0
    wwan0 Link encap:Ethernet HWaddr 0C:5B:8F:27:9A:64

            inet addr:100.68.142.143 Bcast:100.68.142.159 Mask:255.255.255.224
            inet6 addr: fe80::e5b:8fff:fe27:9a64/64 Scope:Link
            UP BROADCAST RUNNING MULTICAST MTU:1500 Metric:1
            RX packets:4734766 errors:0 dropped:0 overruns:0 frame:0
            TX packets:4880322 errors:0 dropped:0 overruns:0 carrier:0
            collisions:0 txqueuelen:1000
            RX bytes:2998220682 (2.7 GiB) TX bytes:1392135460 (1.2 GiB)

jow-, 2015-06-18:
Pastebin the output of "devstatus wwan0"

basinilya, 2015-06-19:
Fine, I admit: it's the 32 bit counter. Please close this ticket.

It's just collectd shows much more traffic, than my ISP charges me, but it's probably because they don't count some packets.

closed 2015-06-19 by jow-