openwrt/packages #1430: alsa-lib: segmentation fault during audio device enumeration process

squeezewrt, 2015-06-18:
After change C library to musl, any access to variable 'local_error' in error.c produce segmentation fault. If remove __thread modifier from variable declaration, segmentation faults do not occur. Is musl not support TLS?

thess, 2015-06-18:
Returning a pointer to static TLS has certainly been problematic in musl if you read the discussion about it. The standard mentions  "use of the pointer returned by this interface after the calling thread has terminated results in undefined behavior". I'm not sure if this relavent to your use of alsa.

I will look into building alsa-lib without __thread since it is an option in configure (compiler check). Stay tuned for more info.

squeezewrt, 2015-06-19:
Not about ALSA:

When you return to work on squeezelite package, you might be interested in
two new patches that I did:
https://github.com/squeezewrt/squeezewrt/tree/master/squeezelite/patches
The first fixes codec loading bug, which I wrote earlier.
The second provides compatibility with musl.

Andrew.

thess, 2015-06-19:
Yes, I figured that out last nite. Listing devices with mpd, aplay, alsamixer, etc. did not have any problems. I was going to debug squeezelite over the weekend, but I guess you got it - thanks.

Close this?

squeezewrt, 2015-06-20:
This error occurs when run squeezelite with -l (List output devices) key.

And yes, You may close this.

thess, 2015-06-20:
Temporary work-around for alsa-lib - see #1443

closed 2015-06-20 by thess