openwrt/packages #1434: Fix davfs2 for musl compiling ?

danielpetre, 2015-06-19:
Can you please fix davfs2 too for musl compiling? thanks!

make[6]: Entering directory '/home/dani/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.10/davfs2-1.5.2/src'
mipsel-openwrt-linux-musl-gcc -DPROGRAM_NAME=\"mount.davfs\" -DDAV_SYS_CONF_DIR=\"/etc/davfs2\" -DDAV_LOCALSTATE_DIR=\"/var/run\" -DDAV_SYS_RUN=\"/var/run/mount.davfs\" -DDAV_SYS_CACHE=\"/var/cache/davfs2\" -DDAV_SECRETS=\"secrets\" -DDAV_CONFIG=\"davfs2.conf\" -DDAV_CERTS_DIR=\"certs\" -DDAV_CLICERTS_DIR=\"private\" -DDAV_DATA_DIR=\"/usr/share/davfs2\" -DLOCALEDIR=\"/usr/share/locale\" -DDAV_USER=\"davfs2\" -DDAV_GROUP=\"davfs2\" -D_FORTIFY_SOURCE=2 -DHAVE_CONFIG_H -I. -I..   -I/home/dani/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.10/usr/include -I/home/dani/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.10/include -I/home/dani/openwrt/staging_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_musl-1.1.10/usr/include -I/home/dani/openwrt/staging_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_musl-1.1.10/include  -Wall -Werror=format-security  -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kec -mdsp -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -I/home/dani/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.10/usr/include  -I/home/dani/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.10/usr/include/neon -MT cache.o -MD -MP -MF .deps/cache.Tpo -c -o cache.o cache.c
cache.c:27:19: fatal error: error.h: No such file or directory
 #include <error.h>
                   ^
compilation terminated.
Makefile:426: recipe for target 'cache.o' failed
make[6]: *** [cache.o] Error 1
make[6]: Leaving directory '/home/dani/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.10/davfs2-1.5.2/src'
Makefile:464: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/home/dani/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.10/davfs2-1.5.2'
Makefile:384: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/dani/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.10/davfs2-1.5.2'
Makefile:69: recipe for target '/home/dani/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.10/davfs2-1.5.2/.built' failed
make[3]: *** [/home/dani/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.10/davfs2-1.5.2/.built] Error 2
make[3]: Leaving directory '/home/dani/openwrt/feeds/packages/net/davfs2'
package/Makefile:191: recipe for target 'package/feeds/packages/davfs2/compile' failed
make[2]: *** [package/feeds/packages/davfs2/compile] Error 2
make[2]: Leaving directory '/home/dani/openwrt'
package/Makefile:188: recipe for target '/home/dani/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.10/stamp/.package_compile' failed
make[1]: *** [/home/dani/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.10/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/dani/openwrt'
/home/dani/openwrt/include/toplevel.mk:181: recipe for target 'world' failed
make: *** [world] Error 2

yousong, 2016-08-19:
Should be fixed by https://github.com/openwrt/packages/commit/2273ff76

closed 2016-08-19 by yousong

danielpetre, 2016-08-19:
It was, thanks, you can close the ticket.