openwrt/packages #1440: Ruby 2.2.2 encounters a segmentation fault with a simple Sinatra script

spemmons, 2015-06-20:
After a fresh image bringing in the full Ruby set of libraries using "menuconfig", I perform "gem install sinatra" and create the following simple script:

```ruby
require 'sinatra'

get '/' do
  'Hello!'
end
```

Then run "ruby script.rb" from an SSH console and in another console submit "curl http://localhost:4567" that receives the expected "Hello" text. The log for the Sinatra script is as follows:

```
[2015-06-20 00:21:03] INFO  WEBrick 1.3.1
[2015-06-20 00:21:03] INFO  ruby 2.2.2 (2015-04-13) [mips-linux-gnu]
== Sinatra (v1.4.6) has taken the stage on 4567 for development with backup from WEBrick
[2015-06-20 00:21:03] INFO  WEBrick::HTTPServer#start: pid=6525 port=4567
127.0.0.1 - - [20/Jun/2015:00:21:07 +0000] "GET / HTTP/1.1" 200 6 0.1402
localhost - - [20/Jun/2015:00:21:07 GMT] "GET / HTTP/1.1" 200 6
- -> /
Segmentation fault
```

I also logged a bug with ruby-lang.org here: https://bugs.ruby-lang.org/issues/11289

Not sure the best place to get traction. Any help would be greatly appreciated...

luizluca, 2015-06-20:
It is probably not Ruby fault. I faced gcc bug before that resulted a non
deterministic dead lock but no segfault until now.
I'll try to reproduce with debug info and I hope while debugging

luizluca, 2015-06-20:
I just tested on x86, trunk using
https://downloads.openwrt.org/snapshots/trunk/x86/generic/openwrt-x86-generic-combined-ext4.img.gz

# opkg update
# opkg install ruby-stdlib
# gem install --no-ri sinatra # it took ages to finish without --no-ri
# cat > sinatra.rb <<EOF
require 'sinatra'
get '/' do
  'Hello!'
end
EOF
# ruby sinatra.rb 
[2015-06-20 02:40:48] INFO  WEBrick 1.3.1
[2015-06-20 02:40:48] INFO  ruby 2.2.2 (2015-04-13) [i486-linux-gnu]
== Sinatra (v1.4.6) has taken the stage on 4567 for development with backup from WEBrick
[2015-06-20 02:40:48] INFO  WEBrick::HTTPServer#start: pid=5794 port=4567

So, it is something specific for your target or for your build or for musl. I guess it is the former.

What is exactly your openwrt version and target?
Is it really a clean build? It happened a recent musl switch. Did it happened before?
The x86 target (that I used to test) are still using uclibc as they are failing to compile using musl.

I'll probably be able to test it on ar71xx only on 2015-06-29.

spemmons, 2015-06-20:
Attached is the config I used. The target is a WZR-600DHP...

Thanks for the help! -SE-

NeoRaider, 2015-06-20:
Might be the same issue I faced in #1390. It's been fixed today in the musl git, but not in OpenWrt yet.

NeoRaider, 2015-06-20:
Try running ruby with strace, if it shows `svr4_mknod` when segfaulting, it's the same issue.

sbyx, 2015-06-22:
Musl updated, fix should be included now.

spemmons, 2015-06-23:
I have not been able to successfully build since the changes were made. It's probably an unrelated issue, but when building "libnl-tiny" it is missing a library called "libssp". I'm not exactly sure where to log that bug, but can do so with some advice/guidance...

sbyx, 2015-06-23:
Please do a make dirclean to remove your toolchain so it is rebuilt. You should also run make menuconfig and save your settings once so that the recent changes are applied.

spemmons, 2015-06-23:
Ah! I did make clean and rebuilt config, but didn't know to do make dirclean. I will do so.

luizluca, 2015-06-30:
As x86 target is fixed for musl, I retested the issue with r46134.
At least for this target, I've got no problem.

```
root@OpenWrt:/# ruby sinatra.rb 
[2015-06-30 02:14:20] INFO  WEBrick 1.3.1
[2015-06-30 02:14:20] INFO  ruby 2.2.2 (2015-04-13) [i486-linux-gnu]
== Sinatra (v1.4.6) has taken the stage on 4567 for development with backup from WEBrick
[2015-06-30 02:14:20] INFO  WEBrick::HTTPServer#start: pid=4161 port=4567
127.0.0.1 - - [30/Jun/2015:02:14:47 +0000] "GET / HTTP/1.1" 200 6 0.0099
localhost - - [30/Jun/2015:02:14:47 UTC] "GET / HTTP/1.1" 200 6
- -> /
127.0.0.1 - - [30/Jun/2015:02:14:50 +0000] "GET / HTTP/1.1" 200 6 0.0007
localhost - - [30/Jun/2015:02:14:50 UTC] "GET / HTTP/1.1" 200 6
```

@spemmons , do you have any news?

luizluca, 2015-08-29:
@spemmons, can I close this?

spemmons, 2015-08-29:
I have not been in a position to retry this and had to move on. I'll close...

closed 2015-08-29 by spemmons