openwrt/packages #1445: collectd missing from snapshot builds

kroimon, 2015-06-21:
Referring to https://dev.openwrt.org/ticket/19911

The collectd packages are currently missing from the nightly snapshot builds at https://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/packages/packages/.

Probably some build errors here? BTW: Are there any public log files of the build bot?

kroimon, 2015-06-21:
I found the [buildbot logs](http://buildbot.openwrt.org:8010/).
According to the [collectd log](http://buildbot.openwrt.org:8010/broken_packages/ar71xx/collectd/compile.txt), there is a dependeny error to the snmp package. I therefore guess that this issue depends on #1435 and will resolve itself once snmp compiles again.

karlp, 2015-08-24:
As net-snmp was been fixed, and I can see collectd packages in the referenced link, I believe this has been fixed.

closed 2015-08-24 by karlp