openwrt/packages #1450: musl breaks python

commodo, 2015-06-22:
[ Move from OpenWRT mailing list ]

From: micke.prag@telldus.se
	
Hi,

I have discovered that python is broken using musl. When a thread created
using the threading library exists, python segfaults.

This is a simple example application showing the issue:

import time
import threading

def test():
        print("Hello")
        time.sleep(2)
        print("world")

t = threading.Thread(target=test)
t.start()

After printing "world" python segfaults.

root@OpenWrt:~# python test.py
Hello
world
Segmentation fault

This has only been tested/verified by me on the ar71xx platform using AR9331. Built using r46106.

richfelker, 2015-06-22:
This is presumably OpenWRT-specific since the test program works fine with musl-linked Python on Alpine Linux.

mickeprag, 2015-06-23:
I have made a clean build on a separate machine and the issue still remains.

I have not been able to reproduce this issue on ramips using RT5350 (VoCore) so it could be ar71xx related.

commodo, 2015-06-23:
I could not reproduce this on an x86 VM.
Will try to allocate an a71xx machine.

mickeprag, 2015-06-23:
I have a core dump from python. Is this to any help?

commodo, 2015-06-23:
awesome; send it to ardeleanalex@gmail.com
thanks

richfelker, 2015-06-23:
Ah, since it's apparently mips-specific and related to thread exit, it's probably the kernel regression we worked around in this commit:

http://git.musl-libc.org/cgit/musl/commit/?id=ce3688eca920aa77549323f84e21f33522397115

richfelker, 2015-06-23:
See also this thread: http://www.openwall.com/lists/musl/2015/06/19/1

mickeprag, 2015-06-24:
r46075: musl: update musl to 2015-06-20
musl 2015-06-20 includes the above commit.

With this commit it seems to work again.

What I don't understand is why it didn't work in my initial post. Maybe musl was still from <r46075 and not rebuilt for some reason?

commodo, 2015-06-24:
that all depends;
the build system is not too reliable when you have to rebuild the toolchain;
especially now with the musl changes, you'd need to do more often

closed 2015-09-14 by commodo