openwrt/packages #1455: trunk compile error. Package etherwake.

qshenx, 2015-06-23:
`make[2]: Entering directory '/home/qshen/n750wrt/feeds/packages/net/etherwake'
mips-openwrt-linux-musl-gcc -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -Wl,-z,now -Wl,-z,relro -D__UCLIBC__ /home/qshen/n750wrt/build_dir/target-mips_34kc_musl-1.1.10/etherwake-1.09.orig/ether-wake.c -o /home/qshen/n750wrt/build_dir/target-mips_34kc_musl-1.1.10/etherwake-1.09.orig/etherwake
In file included from /home/qshen/n750wrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/include/netinet/ether.h:8:0,
                 from /home/qshen/n750wrt/build_dir/target-mips_34kc_musl-1.1.10/etherwake-1.09.orig/ether-wake.c:94:
/home/qshen/n750wrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/include/netinet/if_ether.h:96:8: error: redefinition of 'struct ethhdr'
 struct ethhdr {
        ^
In file included from /home/qshen/n750wrt/build_dir/target-mips_34kc_musl-1.1.10/etherwake-1.09.orig/ether-wake.c:91:0:
/home/qshen/n750wrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/include/linux/if_ether.h:137:8: note: originally defined here
 struct ethhdr {
        ^
make[2]: *** [/home/qshen/n750wrt/build_dir/target-mips_34kc_musl-1.1.10/etherwake-1.09.orig/.built] Error 1
make[2]: Leaving directory '/home/qshen/n750wrt/feeds/packages/net/etherwake'
make[1]: *** [package/feeds/packages/etherwake/compile] Error 2
make[1]: Leaving directory '/home/qshen/n750wrt'
make: *** [package/etherwake/compile] Error 2`

[image]

qshenx, 2015-06-23:
OS: Ubunut server 14.04.2 x64 LTS
Branch: trunk (2015-06-23)
Package: etherwake

Noltari, 2015-06-23:
There's an open PR: https://github.com/openwrt/packages/pull/1387

closed 2015-06-23 by Noltari