openwrt/packages #1462: Question about appropriate location in the OpenWrt filesystem

ejurgensen, 2015-06-24:
I am the maintainer of package (forked-daapd) which creates and uses a sqlite database for settings, media metadata and cache. The normal location of the database is /var/cache/forked-daapd, but in OpenWrt /var is in memory and not persistent through reboots. That doesn't work well in this case. The user can set a different location in the config file, but I would like to have a sensible default location in the config file that comes with the package. Do you guys have a suggestion for a good preset location? Perhaps /usr/share/forked-daapd?

luizluca, 2015-06-25:
@ejurgensen,

I'm no great OpenWRT expert but I'm playing around for some time.

As OpenWRT normally runs on routers with flash storage, writes are
restricted to a minimum in order to save flash lifetime. So, cache and
databases are written to the tmpfs area (/var). This is the way to go. It
is up to the user to setup another location (optimally configuring it on an
external mount point or setting up extroot).

If the settings part is isolated from other configs, maybe it could live in
/etc. If settings is inside the sqlite database, you could create a new
/etc/config/xxx file and implement some conversion from xxx to the sqlite
database on every service start. Any settings change in database will be
lost on every restart but it is better than nothing.
For cache and metadata (if the latter can be generated without losses),
they originally belong to in /var.

/usr/share is not for databases on any Linux, even on desktop ones. This
path is for static data. Databases are at /var/lib and cache at /var/cache.

Regards,

ejurgensen, 2015-06-25:
Thank you for your reply! I had forgotten that setting up extroot actually solves this. It has been a while since I did that myself. Most users of forked-daapd on OpenWrt will indeed have set up extroot, I think, so I guess there is actually no problem. I will be sticking to the current /var/cache, then.

closed 2015-06-25 by ejurgensen