openwrt/packages #1465: zmq: multiple issues

jow-, 2015-06-25:
Commit 3ff4d6dda5072985817fc3df8c689a4cf6e53cfd updated the zeromq version in zmq/Makefile without adjusting the md5sum.

Furthermore that commit introduced the following change in the Makefile:

    ifneq ($(CONFIG_PACKAGE_libsodium),)
      CONFIGURE_ARGS+= --with-libsodium
    else
      CONFIGURE_ARGS+= --without-libsodium
    endif

This is a no-go as it leads to non-deterministic build results. Either unconditionally depend on `+libsodium` and always pass `--with-libsodium` or uncodnitionally disable it with `--without-libsodium` but do not alter the configure args depending on the package selection.

jow-, 2015-06-25:
@srdgame @thess

thess, 2015-06-25:
@srdgame - package doesn't build after fixing md5sum.
If you want to have libsodium be optional, we could make 2 variants.

srdgame, 2015-06-26:
@thess What should I do if i want to keep libsodium as optional?  Is it possible change the condition to detect the selection of "LIBZMQ_CURVEZMQ"?

thess, 2015-06-26:
I think 2 variant packages would be the best solution. It would be easier for people installing individual pre-built packages rather than forcing someone to do a custom build to choose one over the other.

thess, 2015-07-04:
See PR #1538 for proposed solution to these issues.

thess, 2015-07-13:
Closed by #1538

closed 2015-07-13 by thess