openwrt/packages #1473: Pulseaudio segfault

TobleMiner, 2015-06-25:
Pulseaudio crashes as soon as I start it on my HooToo Trip Mate TM-01:

```
root@daringdo:~# pulseaudio -vvv --system
W: [pulseaudio] main.c: Running in system mode, but --disallow-exit not set!
W: [pulseaudio] main.c: Running in system mode, but --disallow-module-loading not set!
N: [pulseaudio] main.c: Running in system mode, forcibly disabling SHM mode!
N: [pulseaudio] main.c: Running in system mode, forcibly disabling exit idle time!
D: [pulseaudio] core-rtclock.c: Timer slack is set to 50 us.
D: [pulseaudio] core-util.c: setpriority() worked.
I: [pulseaudio] core-util.c: Successfully gained nice level -11.
I: [pulseaudio] main.c: Found user 'pulse' (UID 51) and group 'pulse' (GID 51).
I: [pulseaudio] main.c: Successfully changed user to "pulse".
I: [pulseaudio] main.c: This is PulseAudio 6.0
D: [pulseaudio] main.c: Compilation host: mipsel-openwrt-linux-gnu
D: [pulseaudio] main.c: Compilation CFLAGS: -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kec -mdsp -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/tsys/work/openwrt/trunk/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.10/usr/lib/libiconv-stub/include -I/home/tsys/work/openwrt/trunk/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.10/usr/lib/libintl-stub/include -std=gnu99  -Wall -W -Wextra -Wno-long-long -Wno-overlength-strings -Wunsafe-loop-optimizations -Wundef -Wformat=2 -Wlogical-op -Wsign-compare -Wformat-security -Wmissing-include-dirs -Wformat-nonliteral -Wpointer-arith -Winit-self -Wdeclaration-after-statement -Wfloat-equal -Wmissing-prototypes -Wredundant-decls -Wmissing-declarations -Wmissing-noreturn -Wshadow -Wendif-labels -Wcast-align -Wstrict-aliasing -Wwrite-strings -Wno-unused-parameter -ffast-math -fno-common -fdiagnostics-show-option
D: [pulseaudio] main.c: Running on host: Linux mips 3.18.16 #1 Wed Jun 24 09:12:59 UTC 2015
D: [pulseaudio] main.c: Found 1 CPUs.
I: [pulseaudio] main.c: Page size is 4096 bytes
D: [pulseaudio] main.c: Compiled with Valgrind support: no
D: [pulseaudio] main.c: Running in valgrind mode: no
D: [pulseaudio] main.c: Running in VM: no
D: [pulseaudio] main.c: Optimized build: yes
D: [pulseaudio] main.c: FASTPATH defined, only fast path asserts disabled.
I: [pulseaudio] main.c: Machine ID is daringdo.
I: [pulseaudio] main.c: Using runtime directory /var/run/pulse.
I: [pulseaudio] main.c: Using state directory /var/lib/pulse.
I: [pulseaudio] main.c: Using modules directory /usr/lib/pulse-6.0/modules.
I: [pulseaudio] main.c: Running in system mode: yes
W: [pulseaudio] main.c: OK, so you are running PA in system mode. Please note that you most likely shouldn't be doing that.
W: [pulseaudio] main.c: If you do it nonetheless then it's your own fault if things don't work as expected.
W: [pulseaudio] main.c: Please read http://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/WhatIsWrongWithSystemWide/ for an explanation why system mode is usually a bad idea.
W: [pulseaudio] pid.c: Stale PID file, overwriting.
I: [pulseaudio] main.c: Fresh high-resolution timers available! Bon appetit!
D: [pulseaudio] memblock.c: Using private memory pool with 1024 slots of size 4.0 KiB each, total size is 4.0 MiB, maximum usable slot size is 4056
D: [pulseaudio] module.c: Checking for existence of '/usr/lib/pulse-6.0/modules/module-detect.so': success
D: [pulseaudio] module.c: Checking for existence of '/usr/lib/pulse-6.0/modules/module-esound-protocol-unix.so': success
I: [pulseaudio] module.c: Loaded "module-esound-protocol-unix" (index: #0; argument: "").
I: [pulseaudio] module.c: Loaded "module-native-protocol-unix" (index: #1; argument: "").
I: [pulseaudio] module-stream-restore.c: Successfully opened database file '/tmp/lib/pulse/daringdo-stream-volumes'.
I: [pulseaudio] module.c: Loaded "module-stream-restore" (index: #2; argument: "").
I: [pulseaudio] module-device-restore.c: Successfully opened database file '/tmp/lib/pulse/daringdo-device-volumes'.
I: [pulseaudio] module.c: Loaded "module-device-restore" (index: #3; argument: "").
I: [pulseaudio] module.c: Loaded "module-default-device-restore" (index: #4; argument: "").
I: [pulseaudio] module.c: Loaded "module-rescue-streams" (index: #5; argument: "").
D: [pulseaudio] module-always-sink.c: Autoloading null-sink as no other sinks detected.
Segmentation fault
```

I'm using Pulseaudio 6.0 and Chaos Calmer r46118 on MIPS:

```
root@daringdo:~# uname -a
Linux daringdo 3.18.16 #1 Wed Jun 24 09:12:59 UTC 2015 mips GNU/Linux
```

I've already tried to recompile Pulseaudio from source but that didn't help. It keeps crashing.

thess, 2015-07-05:
Was this version using musl or uClibc?

After a quick check of this problem on the __trunk__ with musl, I was unable to reproduce the problem. It may have been either:
* Not enough memory for pulseaudio on your device (needs >9M free)
* Problem is related to static __thread declaration on mips platforms using musl. Fixed in musl patch on 27-Jun in 
http://git.openwrt.org/?p=openwrt.git;a=commitdiff;h=7586be0444157740114a554ec9164a0770e96

I will try it on CC w/ uClibc in a bit.

TobleMiner, 2015-07-05:
Thanks for your answer, I'm currently using musl.

TobleMiner, 2015-07-05:
I just did a sysupgrade and the change to musl seems to have done a good job. PA doesn't chrash on startup anymore. I'll have to do some further testing to make sure it's working properly though.

TobleMiner, 2015-07-05:
It works! Thanks for your help.

closed 2015-07-05 by TobleMiner