openwrt/packages #1478: please update xl2tpd package in 15.05 branch

duvi, 2015-06-26:
Several changes were made since the fork that should be backported, because xl2tpd is not working without them.

Thank you!

puleglot, 2015-06-27:
Confirming this. Just replacing l2tp.sh with a version from the master branch fixes connection issues for me.

puleglot, 2015-06-27:
Update: adding "no_proto_task=1" to the l2tp.sh also fixes this issue. This is just in case you don't want to backport all the changes the master branch.

dangowrt, 2015-06-28:
Once https://github.com/openwrt/packages/issues/1334 is resolved, the best would be to backport trunk's xl2tpd to CC.

ettavolt, 2015-09-21:
FYI: I've tested 5674a5835e9b89b7438917a380f3a6d68528fa3e xl2tpd, and it worked with netifd from 15.05.

yousong, 2015-09-21:
Thanks for the testing.  Before backporting xl2tpd from master branch to for-15.05, I'd like to see if the pull request to upstream xl2tpd for fixing memory corruption (https://github.com/xelerance/xl2tpd/pull/77) can be reviewed and accepted.  I'd hold for a while and do the backport in its current shape anyway if the review process there is going to take a long time.

yousong, 2015-09-26:
Hi, I just backported the xl2tpd version from master branch to `for-15.05`.

closed 2015-09-26 by yousong

duvi, 2015-09-26:
I just upgraded a router to this, and it doesn't connect l2tp anymore.
I also have latest CC on it (r47046).

The previous version with a patched l2tp.sh (e9e2400) from the other thread was working fine.

Now "ifup vpn" simply does nothing.
"ifstatus vpn" shows:
```
root@OpenWrt:~# ifstatus vpn
{
        "up": false,
        "pending": false,
        "available": false,
        "autostart": true,
        "proto": "l2tp",
        "data": {

        },
        "errors": [
                {
                        "subsystem": "interface",
                        "code": "NO_DEVICE"
                }
        ]
}
```
```
config 'interface' 'vpn'
        option 'proto'    'l2tp'
        option 'server'   'myserver'
        option 'username' 'myuser'
        option 'password' 'mypass'
        option 'mtu'      '1410'
        option 'keepalive' '12'
        option 'defaultroute' '0'

config 'route' 'vpnroute'
        option 'interface' 'vpn'
        option 'target'    '192.168.200.0'
        option 'netmask'   '255.255.255.0'
        option 'gateway'   '192.168.200.1'
```

duvi, 2015-09-26:
The "previous version" was in trunk, not CC.

yousong, 2015-09-26:
Hi,

I also encountered the "NO_DEVICE" issue several times after doing `ifup sankuai`.  It seems to me like a race condition in handling host dependency.

Can you change `l2tp.sh` like the following patch does and see if the issue can go away?

```
diff --git a/net/xl2tpd/files/l2tp.sh b/net/xl2tpd/files/l2tp.sh
index 8a931ad..c170ef4 100644
--- a/net/xl2tpd/files/l2tp.sh
+++ b/net/xl2tpd/files/l2tp.sh
@@ -29,7 +29,6 @@ proto_l2tp_setup() {
 	local ip serv_addr server
 	json_get_var server server && {
 		for ip in $(resolveip -t 5 "$server"); do
-			( proto_add_host_dependency "$interface" "$ip" )
 			serv_addr=1
 		done
 	}
```

duvi, 2015-09-26:
After removing this line and rebooting the router, l2tp successfully connected!
Now let's see how it behaves in the long run, how it survives wan reconnections, etc.
Thank you for your efforts!

yousong, 2015-09-26:
I observed the issue just several days ago and still did not yet figure out the exact order of events that could lead to this state.

I'd like to wait for while and see if a proper fix can be achieved before just removing that line from `l2tp.sh`.  Sorry for the delay anyway.

duvi, 2015-10-01:
After a couple of days working suddenly l2tp did not reconnect.
Successful reconnection:
```
Sep 30 09:39:08 __clientip__ minerva: pppd[13020]: No response to 12 echo-requests
Sep 30 09:39:08 __clientip__ minerva: pppd[13020]: Serial link appears to be disconnected.
Sep 30 09:39:08 __clientip__ minerva: pppd[13020]: Connect time 1155.2 minutes.
Sep 30 09:39:08 __clientip__ minerva: pppd[13020]: Sent 0 bytes, received 0 bytes.
Sep 30 09:39:08 __clientip__ minerva: netifd: Network device 'l2tp-vpn' link is down
Sep 30 09:39:08 __clientip__ minerva: syslog: 06[KNL] interface l2tp-vpn deactivated
Sep 30 09:39:08 __clientip__ minerva: syslog: 02[KNL] 192.168.200.14 disappeared from l2tp-vpn
Sep 30 09:39:08 __clientip__ minerva: pppd[13020]: Connection terminated.
Sep 30 09:39:08 __clientip__ minerva: pppd[13020]: Modem hangup
Sep 30 09:39:08 __clientip__ minerva: syslog: 12[KNL] interface l2tp-vpn deleted
Sep 30 09:39:08 __clientip__ minerva: pppd[13020]: Exit.
Sep 30 09:39:08 __clientip__ minerva: xl2tpd[1030]: child_handler : pppd exited for call 41786 with code 15
Sep 30 09:39:08 __clientip__ minerva: xl2tpd[1030]: call_close: Call 61690 to __serverip__ disconnected
Sep 30 09:39:08 __clientip__ minerva: xl2tpd[1030]: Terminating pppd: sending TERM signal to pid 13020
Sep 30 09:39:08 __clientip__ minerva: xl2tpd[1030]: Disconnecting from __serverip__, Local: 15068, Remote: 25699
Sep 30 09:39:08 __clientip__ minerva: xl2tpd[1030]: Connection 25699 closed to __serverip__, port 1701 (Goodbye!)
Sep 30 09:39:08 __clientip__ minerva: netifd: Interface 'vpn' is now down
Sep 30 09:39:08 __clientip__ minerva: netifd: Interface 'vpn' is setting up now
Sep 30 09:39:08 __clientip__ minerva: xl2tpd[1030]: Can not find tunnel 15068 (refhim=0)
Sep 30 09:39:08 __clientip__ minerva: xl2tpd[1030]: network_thread: unable to find call or tunnel to handle packet.  call = 61690, tunnel = 15068 Dumping.
Sep 30 09:39:08 __clientip__ minerva: xl2tpd[1030]: Connecting to host __serverhostname__, port 1701
Sep 30 09:39:10 __clientip__ minerva: xl2tpd[1030]: Connection established to __serverip__, 1701.  Local: 31544, Remote: 42669 (ref=0/0).
Sep 30 09:39:10 __clientip__ minerva: xl2tpd[1030]: Calling on tunnel 31544
Sep 30 09:39:10 __clientip__ minerva: xl2tpd[1030]: check_control: Received out of order control packet on tunnel 42669 (got 0, expected 1)
Sep 30 09:39:10 __clientip__ minerva: xl2tpd[1030]: handle_packet: bad control packet!
Sep 30 09:39:10 __clientip__ minerva: xl2tpd[1030]: check_control: Received out of order control packet on tunnel 42669 (got 0, expected 1)
Sep 30 09:39:10 __clientip__ minerva: xl2tpd[1030]: handle_packet: bad control packet!
Sep 30 09:39:10 __clientip__ minerva: xl2tpd[1030]: Call established with __serverip__, Local: 48396, Remote: 59389, Serial: 15 (ref=0/0)
Sep 30 09:39:10 __clientip__ minerva: xl2tpd[1030]: start_pppd: I'm running:
Sep 30 09:39:10 __clientip__ minerva: xl2tpd[1030]: "/usr/sbin/pppd"
Sep 30 09:39:10 __clientip__ minerva: xl2tpd[1030]: "/dev/pts/0"
Sep 30 09:39:10 __clientip__ minerva: xl2tpd[1030]: "passive"
Sep 30 09:39:10 __clientip__ minerva: xl2tpd[1030]: "nodetach"
Sep 30 09:39:10 __clientip__ minerva: xl2tpd[1030]: ":"
Sep 30 09:39:10 __clientip__ minerva: xl2tpd[1030]: "file"
Sep 30 09:39:10 __clientip__ minerva: xl2tpd[1030]: "/tmp/l2tp/options.vpn"
Sep 30 09:39:10 __clientip__ minerva: pppd[16420]: pppd 2.4.7 started by root, uid 0
Sep 30 09:39:10 __clientip__ minerva: pppd[16420]: Using interface l2tp-vpn
Sep 30 09:39:10 __clientip__ minerva: pppd[16420]: Connect: l2tp-vpn <--> /dev/pts/0
Sep 30 09:39:10 __clientip__ minerva: kernel: [259839.230000] l2tp-vpn: renamed from ppp1
Sep 30 09:39:11 __clientip__ minerva: xl2tpd[1030]: Can not find tunnel 15068 (refhim=0)
Sep 30 09:39:11 __clientip__ minerva: xl2tpd[1030]: network_thread: unable to find call or tunnel to handle packet.  call = 61690, tunnel = 15068 Dumping.
Sep 30 09:39:13 __clientip__ minerva: pppd[16420]: EAP: Identity prompt "Name"
Sep 30 09:39:13 __clientip__ minerva: pppd[16420]: EAP authentication succeeded
Sep 30 09:39:13 __clientip__ minerva: pppd[16420]: local  IP address 192.168.200.14
Sep 30 09:39:13 __clientip__ minerva: pppd[16420]: remote IP address 192.168.200.1
Sep 30 09:39:13 __clientip__ minerva: syslog: 05[KNL] 192.168.200.14 appeared on l2tp-vpn
Sep 30 09:39:13 __clientip__ minerva: syslog: 06[KNL] 192.168.200.14 disappeared from l2tp-vpn
Sep 30 09:39:13 __clientip__ minerva: syslog: 10[KNL] 192.168.200.14 appeared on l2tp-vpn
Sep 30 09:39:13 __clientip__ minerva: syslog: 03[KNL] interface l2tp-vpn activated
Sep 30 09:39:13 __clientip__ minerva: netifd: Network device 'l2tp-vpn' link is up
Sep 30 09:39:13 __clientip__ minerva: netifd: Interface 'vpn' is now up
```

Failed reconnection:
```
Sep 30 11:14:14 __clientip__ minerva: pppd[16420]: No response to 12 echo-requests
Sep 30 11:14:14 __clientip__ minerva: pppd[16420]: Serial link appears to be disconnected.
Sep 30 11:14:14 __clientip__ minerva: pppd[16420]: Connect time 95.1 minutes.
Sep 30 11:14:14 __clientip__ minerva: pppd[16420]: Sent 0 bytes, received 0 bytes.
Sep 30 11:14:14 __clientip__ minerva: netifd: Network device 'l2tp-vpn' link is down
Sep 30 11:14:14 __clientip__ minerva: syslog: 05[KNL] interface l2tp-vpn deactivated
Sep 30 11:14:14 __clientip__ minerva: syslog: 06[KNL] 192.168.200.14 disappeared from l2tp-vpn
Sep 30 11:14:14 __clientip__ minerva: pppd[16420]: Connection terminated.
[... cut ...]
Sep 30 11:14:26 __clientip__ minerva: xl2tpd[1030]: Can not find tunnel 19977 (refhim=0)
Sep 30 11:14:26 __clientip__ minerva: xl2tpd[1030]: network_thread: unable to find call or tunnel to handle packet.  call = 0, tunnel = 19977 Dumping.
Sep 30 11:14:27 __clientip__ minerva: xl2tpd[1030]: Can not find tunnel 9063 (refhim=0)
Sep 30 11:14:27 __clientip__ minerva: xl2tpd[1030]: network_thread: unable to find call or tunnel to handle packet.  call = 0, tunnel = 9063 Dumping.
```
And then it all stops here.

yousong, 2015-10-02:
No idea what's going wrong there.  I think the following information can be helpful if they can be provided.

- Is the l2tp interface the main internet connection for the device?
- Do you have an option like `checkup_interval 20` for the l2tp interface?  The option is mainly for l2tp protocol handling and not yet in the wiki docs.
- When the interface stopped retrying connection, could it be brought up with `ifup` command and if not what was the output of `ifstatus` command?

Thanks for reporting the issue.  I am about to update xl2tpd to include several memory corruption bugfix.

duvi, 2015-10-02:
1. No, there is a pppoe wan connection as main internet connection. It did not reconnect at that specific time though, it has been up for 4 days since the router started.
1.b. l2tp runs over ipsec, which rekeys every 30-40 minutes. Sometimes the l2tp connection survives this rekeying, sometimes it doesn't, then it ends with "No response to 12 echo-requests" and tries to reconnect. In 4 days of uptime, this has happened around 15 times, it seems totally random when it doesn't survive, I can't find any pattern in the frequency. At Sep 30 11:14:14, it failed to reconnect.
2. I only use the settings that are in the config file in my comment 6 days ago.
3. I noticed 12 hours later that the interface has been down. "Ifup vpn" did immediately reconnect.

Thanks for your efforts!

yousong, 2015-10-02:
Hmm, that `checkup_interval` option can be leveraged here.  It's mainly for the case that xl2tpd received the connect request from netifd yet failed to complete it.  Netifd will test if the interface is up after `checkup_interval` seconds since the last setup attempt and retry otherwise.

duvi, 2015-10-02:
You mean I should try to set checkup_interval to 20?

yousong, 2015-10-02:
Set it to a value that is sufficient for completing the L2TP connection.  20 seconds should be enough for most cases I think.

duvi, 2015-10-02:
Thanks, I'll try it.

Do you have any idea why the l2tp connection is dropped randomly after ipsec rekeying? Or how I could avoid it?

yousong, 2015-10-02:
Sorry, man, never encountered that before.

duvi, 2015-10-02:
All right, thanks!

yousong, 2016-12-01:
Hi, @cbabs , this issue was opened because xl2tpd was nearly not usable at that time and it was now resolved in this regard.  Please open a new one describing what's the current issue and ideally your intended plan.