openwrt/packages #1497: bluez: bluez 5 bluetoothctl does not work with dbus policy packaged

PolarSolvent, 2015-06-28:
`<policy at_console="true">` I don't know if at_console is supported in openwrt, but it's impractical for devices where there usually is no console.
bluez5 also [changed](www.bluez.org/bluez-5-api-introduction-and-porting-guide/) a lot of interface names.

The policy as shipped produces errors like: 
`Sun Jun 28 01:08:28 2015 auth.notice dbus[628]: [system] Rejected send message, 1 matched rules; type="method_call", sender=":1.1" (uid=0 pid=871 comm="bluetoothctl ") interface="org.freedesktop.DBus.ObjectManager" member="GetManagedObjects" error name="(unset)" requested_reply="0" destination="org.bluez" (uid=0 pid=638 comm="/usr/bin/bluetoothd -n ")`

The policy delivered with bluez sources in `bluez-5.30/src/bluetooth.conf` is probably a better fit since it has correct interfacenames and does not use the at_console policy. Putting it in /etc/dbus-1/system.d/bluetooth.conf lets you pair devices with bluetoothctl
```
[bluetooth]# scan on
[bluetooth]# agent on
[bluetooth]# default-agent
[bluetooth]# pair 00:1F:81:00:02:00
Attempting to pair with 00:1F:81:00:02:00
[CHG] Device 00:1F:81:00:02:00 Connected: yes
Request PIN code
[agent] Enter PIN code: 0000
[CHG] Device 00:1F:81:00:02:00 Paired: yes
Pairing successful
```

PolarSolvent, 2015-07-25:
Something like this might be a solution
```diff
diff --git a/utils/bluez/Makefile b/utils/bluez/Makefile
index ebc1646..363bf2c 100644
--- a/utils/bluez/Makefile
+++ b/utils/bluez/Makefile
@@ -142,7 +142,7 @@ define Package/bluez-daemon/install
        $(INSTALL_DIR) $(1)/etc/config
        $(INSTALL_DATA) ./files/bluetooth.config $(1)/etc/config/bluetooth
        $(INSTALL_DIR) $(1)/etc/dbus-1/system.d/
-       $(INSTALL_DATA) ./files/bluetooth.dbus $(1)/etc/dbus-1/system.d/bluetooth.conf
+       $(INSTALL_DATA) $(PKG_BUILD_DIR)/src/bluetooth.conf $(1)/etc/dbus-1/system.d/bluetooth.conf
        $(INSTALL_DIR) $(1)/etc/bluetooth
        $(INSTALL_DATA) $(PKG_BUILD_DIR)/src/main.conf $(1)/etc/bluetooth/main.conf
        $(INSTALL_DATA) $(PKG_BUILD_DIR)/profiles/network/network.conf $(1)/etc/bluetooth/network.conf
```

RogerLaw, 2017-02-05:
this solved my problem

ecammit, 2018-03-13:
This also fixed my issue.

closed 2018-06-10 by hnyman