openwrt/packages #1502: mwan3: wrong interface status

joaochainho, 2015-06-29:
Hi,
I found that mwan3 (1.6-1) is getting the status off all interfaces wrong. Status is always 'unknown'.
Tested with r46120/musl, but r46020/uClibc works fine.
Looking into the code, I found that some 'uci' commands are malformed. So I'd like to suggest a patch to fix it.
```patch
diff --git a/net/mwan3/files/usr/sbin/mwan3 b/net/mwan3/files/usr/sbin/mwan3
index c69d94e..b887fa5 100755
--- a/net/mwan3/files/usr/sbin/mwan3
+++ b/net/mwan3/files/usr/sbin/mwan3
@@ -58,7 +58,7 @@ ifup()
 
        config_get enabled "$1" enabled 0
 
-       device=$(uci get -p /var/state network.$1.ifname) &> /dev/null
+       device=$(uci -p /var/state get network.$1.ifname) &> /dev/null
 
        if [ -n "$device" ] ; then
                [ "$enabled" -eq 1 ] && ACTION=ifup INTERFACE=$1 DEVICE=$device /sbin/hotplug-call iface
@@ -76,7 +76,7 @@ interfaces()
        check_iface_status()
        {
                let iface_id++
-               device=$(uci get -p /var/state network.$1.ifname) &> /dev/null
+               device=$(uci -p /var/state get network.$1.ifname) &> /dev/null
 
                if [ -z "$device" ]; then
                        echo " interface $1 is unknown"
@@ -92,7 +92,7 @@ interfaces()
                fi
 
                if [ -n "$($IP rule | awk '$5 == "'$device'"')" -a -n "$($IPT -S mwan3_iface_$1 2> /dev/null)" -a -n "$($IP route list table $iface_id default dev $device 2> /dev/null)" ]; then
-                       if [ -n "$(uci get -p /var/state mwan3.$1.track_ip 2> /dev/null)" ]; then
+                       if [ -n "$(uci -p /var/state get mwan3.$1.track_ip 2> /dev/null)" ]; then
                                echo " interface $1 is online (tracking $tracking)"
                        else
                                echo " interface $1 is online"
@@ -101,7 +101,7 @@ interfaces()
                        echo " interface $1 error"
                else
                        if [ "$enabled" -eq 1 ]; then
-                               if [ -n "$(uci get -p /var/state mwan3.$1.track_ip 2> /dev/null)" ]; then
+                               if [ -n "$(uci -p /var/state get mwan3.$1.track_ip 2> /dev/null)" ]; then
                                        echo " interface $1 is offline (tracking $tracking)"
                                else
                                        echo " interface $1 is offline"
```
Hope it's useful.

Adze1502, 2015-07-02:
Thnx for reporting. Will change it asap.

arfett, 2015-07-06:
mwan3-luci:
https://github.com/openwrt/packages/pull/1546

closed 2015-07-27 by champtar