openwrt/packages #1517: Add optional mdns support to the p910nd package

gdamjan, 2015-07-01:
Anouncing the printer attached to an openwrt device with Zeroconf (Bonjour/Avahi) wastly simplifies its setup on the clients. This is documented here http://wiki.openwrt.org/doc/howto/p910nd.server for Avahi.

Now, the same would be beneficial with mdns too. What would be required is, a config item for the printer description, and a change to the init script (similar to what dropbear has) to optionally announce the printer with mdns (if available).

pkerling, 2016-04-02:
Thank you for the suggestion. I'm looking into it, but I have trouble getting mdnsd to work even with dropbear... Do you happen to have some documentation about how this should work?

gdamjan, 2016-04-03:
seems this is what dropbear does
https://github.com/openwrt/openwrt/blob/master/package/network/services/dropbear/files/dropbear.init#L81

pkerling, 2016-04-04:
I've seen that - but it does nothing for me (mdns is installed and running).
Does it actually work for you with dropbear? Can you get me the output of `avahi-browse -a -v -t` on a client machine?

gdamjan, 2016-04-04:
huh
this did work, but now it doesn't on my client machine (Arch Linux).
Resolving works, but I'm not seeing the dropbear service. Openwrt is latest 15.05 self compiled from git

pkerling, 2016-04-05:
I'm afraid I won't be able to implement this until mdns has some sort of documentation and it works with dropbear or some other service :(

diizzyy, 2017-06-07:
@pkerling 
Can we fix this or are we still missing docs/functionality in mdns?

pkerling, 2017-06-07:
I'm on LEDE at the moment. Seems mdns was [renamed](http://lists.infradead.org/pipermail/lede-dev/2017-February/005881.html) to umdns in LEDE, which works for me. The mdns package in LEDE does not - ymmv with OpenWRT.

Please test 1ac617043b630f02aa4a53bb643a39028eed6b0b and report if it works for you. You can just copy the config and init files, you don't have to rebuild the package. Note that you will also have to apply openwrt/openwrt#490 if any mdns string contain spaces.

diizzyy, 2018-06-11:
@hnyman @thess 
Solved, please close

pkerling, 2018-06-11:
@diizzyy Did you test 1ac6170 or did you solve it another way?

diizzyy, 2018-06-11:
No, I didn't. Since it was comitted I assumed that someone else tested it (and there were no other comments added). Sorry for the misunderstanding.

pkerling, 2018-06-11:
I tested myself, but would be better to have another person test it. Would you be able to do that so I can merge?

closed 2018-08-12 by thess

corecode, 2018-11-27:
the mdns support commit seems to be dangling in the repo; certainly it's not in master.  could you please cherry pick it into master?

pkerling, 2018-11-27:
I don't think anyone tested it yet

corecode, 2018-11-29:
I applied the additional code to the init file, and I can confirm that it works.