openwrt/packages #1519: latest patch/version squid 3.5.5 build/patching/? failed

build000, 2015-07-01:
make[3]: Entering directory '/magazyn/Pliki/openwrt/trunk/feeds/packages/net/squid'
. /magazyn/Pliki/openwrt/trunk/include/shell.sh; xzcat /magazyn/Pliki/openwrt/trunk/dl/squid-3.5.5.tar.xz | tar -C /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/squid-3.5.5/.. -xf - 

Applying ./patches/001-cross_compile.patch using plaintext: 
patching file src/Makefile.in
patching file configure

Applying ./patches/100-mime.patch using plaintext: 
patching file src/mime.conf.default

Applying ./patches/200-musl-compat.patch using plaintext: 
patching file lib/rfcnb/rfcnb-io.c
Hunk #1 FAILED at 36.
1 out of 1 hunk FAILED -- saving rejects to file lib/rfcnb/rfcnb-io.c.rej
Patch failed!  Please fix ./patches/200-musl-compat.patch!
Makefile:131: recipe for target '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/squid-3.5.5/.prepared_427e3c2dc6278405ce17c0105699f593' failed
make[3]: *** [/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/squid-3.5.5/.prepared_427e3c2dc6278405ce17c0105699f593] Error 1
make[3]: Leaving directory '/magazyn/Pliki/openwrt/trunk/feeds/packages/net/squid'
package/Makefile:191: recipe for target 'package/feeds/packages/squid/compile' failed
make[2]: *** [package/feeds/packages/squid/compile] Error 2
make[2]: Leaving directory '/magazyn/Pliki/openwrt/trunk'
package/Makefile:188: recipe for target '/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/stamp/.package_compile' failed
make[1]: *** [/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/stamp/.package_compile] Error 2
make[1]: Leaving directory '/magazyn/Pliki/openwrt/trunk'
/magazyn/Pliki/openwrt/trunk/include/toplevel.mk:181: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2

build000, 2015-07-03:
https://github.com/openwrt/packages/commit/2f3ab73ffaeb482327651d02997a0125f012ffa1
fixing this problem - pleas close this issue.

yinuo2010, 2015-07-05:
I meet the same problem, Ipc::Atomic::WordT<unsigned long long>::get() const': MemStore.cc:(.text._ZNK3Ipc6Atomic5WordTIyE3getEv[_ZNK3Ipc6Atomic5WordTIyE3getEv]+0x4): undefined reference to__sync_fetch_and_add_8', anyone has the solution?

finasfinas, 2015-07-08:
make[7]: Leaving directory `/home/finas/openwrt.C5.musl/build_dir/target-mips_34kc_musl-1.1.10/squid-3.5.5/src/adaptation'
make[7]: Entering directory `/home/finas/openwrt.C5.musl/build_dir/target-mips_34kc_musl-1.1.10/squid-3.5.5/src'
/bin/sh ../libtool  --tag=CXX   --mode=link mips-openwrt-linux-musl-g++ -Wall -Wpointer-arith -Wwrite-strings -Wcomments -Wshadow -Werror -pipe -D_REENTRANT -I/home/finas/openwrt.C5.musl/staging_dir/target-mips_34kc_musl-1.1.10/usr/include  -I/home/finas/openwrt.C5.musl/staging_dir/target-mips_34kc_musl-1.1.10/usr/include  -Os -pipe -mno-branch-likely -mips32r2 -mtune=74kc -mdspr2 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -std=c++11 -export-dynamic -dlopen force -L/home/finas/openwrt.C5.musl/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib -L/home/finas/openwrt.C5.musl/staging_dir/target-mips_34kc_musl-1.1.10/lib -L/home/finas/openwrt.C5.musl/staging_dir/toolchain-mips_34kc_gcc-4.9-linaro_musl-1.1.10/usr/lib -L/home/finas/openwrt.C5.musl/staging_dir/toolchain-mips_34kc_gcc-4.9-linaro_musl-1.1.10/lib -znow -zrelro  -o squid AclRegs.o AuthReg.o AccessLogEntry.o AsyncEngine.o YesNoNone.o cache_cf.o CacheDigest.o cache_manager.o carp.o cbdata.o ChunkedCodingParser.o client_db.o client_side.o client_side_reply.o client_side_request.o BodyPipe.o clientStream.o CollapsedForwarding.o CompletionDispatcher.o ConfigOption.o ConfigParser.o CpuAffinity.o CpuAffinityMap.o CpuAffinitySet.o debug.o delay_pools.o DelayId.o DelayBucket.o DelayConfig.o DelayPool.o DelaySpec.o DelayTagged.o DelayUser.o DelayVector.o NullDelayId.o ClientDelayConfig.o disk.o DiskIO/DiskIOModule.o DiskIO/ReadRequest.o DiskIO/WriteRequest.o dlink.o dns_internal.o DnsLookupDetails.o errorpage.o ETag.o event.o EventLoop.o external_acl.o ExternalACLEntry.o FadingCounter.o fatal.o fd.o fde.o filemap.o fqdncache.o FwdState.o gopher.o helper.o htcp.o http.o HttpHdrCc.o HttpHdrRange.o HttpHdrSc.o HttpHdrScTarget.o HttpHdrContRange.o HttpHeader.o HttpHeaderTools.o HttpBody.o HttpMsg.o HttpParser.o HttpReply.o RequestFlags.o HttpRequest.o HttpRequestMethod.o icp_v2.o icp_v3.o int.o internal.o ipc.o ipcache.o  SquidList.o main.o MasterXaction.o mem.o mem_node.o MemBuf.o MemObject.o mime.o mime_header.o multicast.o neighbors.o Notes.o Packer.o Parsing.o  pconn.o peer_digest.o peer_proxy_negotiate_auth.o peer_select.o peer_sourcehash.o peer_userhash.o PeerPoolMgr.o redirect.o refresh.o RemovalPolicy.o send-announce.o MemBlob.o SBuf.o SBufExceptions.o SBufDetailedStats.o SBufStatsAction.o  SquidMath.o SquidNew.o stat.o StatCounters.o StatHist.o String.o StrList.o stmem.o store.o StoreFileSystem.o store_io.o StoreIOState.o store_client.o store_digest.o store_dir.o store_key_md5.o store_log.o store_rebuild.o store_swapin.o store_swapmeta.o store_swapout.o StoreMetaUnpacker.o StoreMeta.o StoreMetaMD5.o StoreMetaSTD.o StoreMetaSTDLFS.o StoreMetaURL.o StoreMetaVary.o StoreStats.o StoreSwapLogData.o SwapDir.o Transients.o MemStore.o time.o tools.o tunnel.o  url.o urn.o wccp.o wccp2.o whois.o wordlist.o   LoadableModule.o LoadableModules.o DiskIO/DiskIOModules_gen.o err_type.o err_detail_type.o globals.o hier_code.o icp_opcode.o LogTags.o lookup_t.o repl_modules.o swap_log_op.o auth/libacls.la ident/libident.la acl/libacls.la acl/libstate.la auth/libauth.la libAIO.a libBlocking.a libDiskDaemon.a libDiskThreads.a libIpcIo.a libMmapped.a acl/libapi.la base/libbase.la libsquid.la ip/libip.la fs/libfs.la ssl/libsslsquid.la ssl/libsslutil.la ipc/libipc.la mgr/libmgr.la anyp/libanyp.la comm/libcomm.la eui/libeui.la helper/libhelper.la http/libsquid-http.la icmp/libicmp.la icmp/libicmp-core.la log/liblog.la format/libformat.la clients/libclients.la servers/libservers.la ftp/libftp.la  DiskIO/AIO/AIODiskIOModule.o DiskIO/Blocking/BlockingDiskIOModule.o DiskIO/DiskDaemon/DiskDaemonDiskIOModule.o DiskIO/DiskThreads/DiskThreadsDiskIOModule.o DiskIO/IpcIo/IpcIoDiskIOModule.o DiskIO/Mmapped/MmappedDiskIOModule.o repl/liblru.a -lrt -lpthread  -lcrypt  adaptation/libadaptation.la   parser/libsquid-parser.la ../lib/libmisccontainers.la ../lib/libmiscencoding.la ../lib/libmiscutil.la -L/home/finas/openwrt.C5.musl/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib -L/home/finas/openwrt.C5.musl/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib -lssl -lcrypto   -L/home/finas/openwrt.C5.musl/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib -lgnutls      ../compat/libcompat-squid.la  -ldl -L.. -lltdl 
libtool: link: rm -f .libs/squid.nm .libs/squid.nmS .libs/squid.nmT
libtool: link: (cd .libs && mips-openwrt-linux-musl-gcc -Os -pipe -mno-branch-likely -mips32r2 -mtune=74kc -mdspr2 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -c -fno-builtin "squidS.c")
libtool: link: rm -f ".libs/squidS.c" ".libs/squid.nm" ".libs/squid.nmS" ".libs/squid.nmT"
libtool: link: mips-openwrt-linux-musl-g++ -Wall -Wpointer-arith -Wwrite-strings -Wcomments -Wshadow -Werror -pipe -D_REENTRANT -I/home/finas/openwrt.C5.musl/staging_dir/target-mips_34kc_musl-1.1.10/usr/include -I/home/finas/openwrt.C5.musl/staging_dir/target-mips_34kc_musl-1.1.10/usr/include -Os -pipe -mno-branch-likely -mips32r2 -mtune=74kc -mdspr2 -fno-caller-saves -fhonour-copts -Wno-error=unuse [... cut ...]
make[3]: *** [/home/finas/openwrt.C5.musl/build_dir/target-mips_34kc_musl-1.1.10/squid-3.5.5/.built] Error 2
make[3]: Leaving directory `/home/finas/openwrt.C5.musl/feeds/packages/net/squid'
make[2]: *** [package/feeds/packages/squid/compile] Error 2
make[2]: Leaving directory `/home/finas/openwrt.C5.musl'
make[1]: *** [/home/finas/openwrt.C5.musl/staging_dir/target-mips_34kc_musl-1.1.10/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/finas/openwrt.C5.musl'
make: *** [world] Error 2

same here. trying to use squid luci package and it's not working.

xxiao, 2015-11-08:
Same errors on CC release

xxiao, 2015-11-08:
"fixed" it for me:

xxiao, 2015-11-08:
cat patches/200-atomic-squid.patch 
diff -urpN squid-3.5.11/src/ipc/AtomicWord.h squid-3.5.11.new/src/ipc/AtomicWord.h
--- squid-3.5.11/src/ipc/AtomicWord.h	2015-11-01 04:44:25.000000000 -0600
+++ squid-3.5.11.new/src/ipc/AtomicWord.h	2015-11-08 11:38:01.730681073 -0600
@@ -18,7 +18,7 @@ namespace Atomic
 /// Whether atomic operations support is available
 bool Enabled();
 
-#if HAVE_ATOMIC_OPS
+#if 0 //HAVE_ATOMIC_OPS
 
 /// Supplies atomic operations for an integral Value in memory shared by kids.
 /// Used to implement non-blocking shared locks, queues, tables, and pools.

build000, 2015-11-09:
generally fixed to trunk

closed 2015-11-09 by build000