openwrt/packages #1521: shadow-su: Cannot drop the controlling terminal

pjoe, 2015-07-02:
Found when trying to run postgres on 15.05RC2, but simple to reproduce with just:
```
# su nobody -c ls
su: Cannot drop the controlling terminal
```

This is pretty bad as it prevents all init scripts and so on from doing su (and in particular postgres from starting)

pjoe, 2015-07-02:
FWIW: I managed to use start-stop-daemon in the pgsql init script to work around this using something like:
```
start-stop-daemon -S -c postgres -x /usr/bin/postmaster >> ${pglog} 2>&1 -- -D "${pgdata}" &

start-stop-daemon -S -c postgres -x "${pgctl}" -- stop -D "${pgdata}" -s -m fast
```

dangowrt, 2015-07-02:
See also https://github.com/openwrt/packages/issues/1282#issuecomment-116904012

pjoe, 2015-07-02:
after adding some debug output, I can see that what happens is that the call to open /dev/tty fails with errno=13 (EACCESS). Permisiions for /dev/tty is only for set for only root to rw.

pjoe, 2015-07-02:
and after giving everyone access (chmod a+rw /dev/tty) then the ioctl call fails with errno=25 (ENOTTY)

pjoe, 2015-07-02:
following patch handle those cases:
```
--- a/src/su.c
+++ b/src/su.c
@@ -1090,8 +1090,12 @@
 
 		if (fd >= 0) {
 			err = ioctl (fd, TIOCNOTTY, (char *) 0);
+			if (-1 == err && ENOTTY == errno) {
+				/* There are no controlling terminal already */
+				err = 0;
+			}
 			(void) close (fd);
-		} else if (ENXIO == errno) {
+		} else if (ENXIO == errno || EACCES == errno) {
 			/* There are no controlling terminal already */
 			err = 0;
 		}
```

thess, 2015-07-04:
Closed by #1532 (re-open if I'm wrong)

closed 2015-07-04 by thess