openwrt/packages #1522: gammu not build - full language support (and musl ?)

build000, 2015-07-02:
make[3]: Entering directory '/magazyn/Pliki/openwrt/trunk/feeds/packages/utils/gammu'
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/libiconv-full/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/libintl-full/include  -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/usr/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/include/fortify -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/libiconv-full/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/libintl-full/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/libiconv-full/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/libintl-full/include  -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/usr/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/include/fortify -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/libiconv-full/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/libintl-full/include " LDFLAGS="-L/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/usr/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/lib -znow -zrelro -L/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/libiconv-full/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/libintl-full/lib " make -j1 -C /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0/. AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/libiconv-full/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/libintl-full/include" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CROSS="mips-openwrt-linux-musl-" ARCH="mips" ;
make[4]: Entering directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
make[5]: Entering directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
make[6]: Entering directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
make[6]: Leaving directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
[  0%] Built target string
make[6]: Entering directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
make[6]: Leaving directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
[ 23%] Built target libGammu
make[6]: Entering directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
make[6]: Leaving directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
[ 23%] Built target array
make[6]: Entering directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
make[6]: Leaving directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
[ 23%] Built target printing
make[6]: Entering directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
make[6]: Leaving directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
[ 24%] Built target cmdline
make[6]: Entering directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
make[6]: Leaving directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
[ 24%] Built target memorydisplay
make[6]: Entering directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
make[6]: Leaving directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
[ 25%] Built target messagecmdline
make[6]: Entering directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
make[6]: Leaving directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
[ 25%] Built target messagedisplay
make[6]: Entering directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
make[6]: Leaving directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
[ 26%] Built target array-test
make[6]: Entering directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
make[6]: Leaving directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
make[6]: Entering directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/gammu-1.34.0'
Linking C executable at-ccfc-reply
/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/lib/gcc/mips-openwrt-linux-musl/4.8.3/../../../../mips-openwrt-linux-musl/bin/ld: warning: libiconv.so.2, needed by /magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/libglib-2.0.so, not found (try using -rpath or -rpath-link)
/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/lib/gcc/mips-openwrt-linux-musl/4.8.3/../../../../mips-openwrt-linux-musl/bin/ld: warning: libintl.so.8, needed by /magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/libglib-2.0.so, not found (try using -rpath or -rpath-link)
../libgammu/libGammu.so.7.1.34.0: undefined reference to `libiconv'
../libgammu/libGammu.so.7.1.34.0: undefined reference to `libiconv_open'
/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/libglib-2.0.so: undefined reference to `libintl_bind_textdomain_codeset'
../libgammu/libGammu.so.7.1.34.0: undefined reference to `libiconv_close'
/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/libglib-2.0.so: undefined reference to `libintl_dcgettext'
/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/libglib-2.0.so: undefined reference to `libintl_textdomain'
../libgammu/libGammu.so.7.1.34.0: undefined reference to `libintl_bindtextdomain'
/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/libglib-2.0.so: undefined reference to `libintl_gettext'
../libgammu/libGammu.so.7.1.34.0: undefined reference to `libintl_dgettext'
[... cut ...]
make[2]: Leaving directory '/magazyn/Pliki/openwrt/trunk'
package/Makefile:188: recipe for target '/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/stamp/.package_compile' failed
make[1]: *** [/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/stamp/.package_compile] Error 2
make[1]: Leaving directory '/magazyn/Pliki/openwrt/trunk'
/magazyn/Pliki/openwrt/trunk/include/toplevel.mk:181: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2

KanjiMonster, 2015-08-20:
@aTanW

aTanW, 2015-08-20:
Confirmed.
Working on it...

aTanW, 2015-08-21:
Patch in queue

build000, 2015-08-26:
https://github.com/openwrt/packages/commit/9cad7e86a15143f976e95cb86b751ce8f47edd34 - problem fixing.
gammu built up and working properly - obviously based on all the constraints gettext openwrt.

closed 2015-08-26 by build000

adibhannachi, 2016-03-14:
I have the same problem, but this patch doesn't solve the problem.
---
usr/lib/libglib-2.0.so: undefined reference to `libintl_dngettext'
collect2: ld returned 1 exit status
make[5]: *** [smsd/gammu-smsd] Error 1