openwrt/packages #1535: knxd: configure fails for problem with argp-standalone

tru7, 2015-07-03:
I could need some help from the experts. The package knxd compiles fine on Buildbot for all plattforms but mpc85xx and x86. But I have no idea why because it compiles fine in my own toolchain for mpc85xx. I think it must be related to the move of argp-standalone from packages.git to the core a couple of weeks ago. Before that move of argp-standalone it compiled well on all plattforms.
The error in compile.txt says:
checking argp.h usability... yes
checking argp.h presence... yes
checking for argp.h... yes
checking for library containing argp_parse... no
configure: error: argp_parse not found

The configure.ac has two lines for that:
AC_CHECK_HEADER(argp.h,,[AC_MSG_ERROR([argp_parse not found])])
AC_SEARCH_LIBS(argp_parse,argp,,[AC_MSG_ERROR([argp_parse not found])])

This makes two attempts to compile conftest.c. The first attempt fails and the second one succeeds by adding a "-largp" to the command according to the config.log on my buildroot. I have no clue why this fails on Buildbot for just two plattforms. I must assume there is a problem on Buildbot when creating the InstallDev but I have no access to logs there.

I would place additional arguments in Makefile but found no way to put "-larp" to the end of the test-command line, the parameter is always placed before the conftest.c while it's needed after to compile successfully.

Could it help to patch the line in configure.ac somehow? How would this look like?

thess, 2015-07-03:
I don't think it has anything to do with moving the package -- More to do with MUSL (or the definition of _LIBC). I look into this shortly.

tru7, 2015-07-03:
Perfect, thanks for your quick feedback.

tru7, 2015-07-04:
I just found there is a logs.tar.gz on Buildbot containing all config.log files. Now I see why configure fails. There are several lines saying:
argp-parse.c: undefined reference to `__stack_chk_fail_local
It's probably easy for an expert to tell what this means. In the meantime I may try to find out myself but I'm afraid that is beyond my horizon unless Google points directly to the solution.

tru7, 2015-07-04:
I read at some places that this error happens if ld is used instead of gcc as linker. It seems ld is used indeed, from knxd config.log on Buildbot:
/mnt/dl/slave/mpc85xx/build/staging_dir/toolchain-powerpc_8540_gcc-4.8-linaro_musl-1.1.10/lib/gcc/powerpc-openwrt-linux-muslspe/4.8.3/../../../../powerpc-openwrt-linux-muslspe/bin/ld: conftest: hidden symbol `__stack_chk_fail_local' isn't defined
Would it help to define gcc as linker and how is the common way to do that?
I still don't understand why it works in my own toolchain for mpc85xx. The check for -largp is absolutely identical and works fine while it throws this error on Buildbot.

thess, 2015-07-04:
I think the real issue is load order and the supression of default libs. Unfortunately, I needed to add libstdcpp as a dependency as well. Maybe a better fix is possible, but I didn't find it.

tru7, 2015-07-05:
I thought of load order as well. If guess if there was a -lc after the -largp in the test,  that __stack_chk_fail_local could be resolved. I also thought of just dropping the test and just provide -largp in the config, but that would be nasty.

If I could just see the problem in my own toolchain I could try out myself. Do you have any hint for differences on Buildbot compared to mine? I think a just use standard options.

Anyway, thank you very much for your efforts. I just wish I could work on that myself.

tru7, 2015-07-05:
How about using:
AC_SEARCH_LIBS(argp_parse,argp,,[AC_MSG_ERROR([argp_parse not found])],-lc)
instead of:
AC_SEARCH_LIBS(argp_parse,argp,,[AC_MSG_ERROR([argp_parse not found])])
assuming -lc is need to resolve the reference for __stack_chk_fail_local in the check?

sbyx, 2015-07-05:
For musl we automatically add -lssp_nonshared which provides __stack_chk_fail_local, however this is done  by GCC implicitly for all compilations.

thess, 2015-07-05:
I think the -nodefaultlibs in LDFLAGS was the culprit (at least in this case)

thess, 2015-07-05:
Closed by e470ae5ccfeab42c1e401fb5a1994431b06b8f36

closed 2015-07-05 by thess

tru7, 2015-07-05:
@sbyx any hint why my mpc85xx toolchain r46160 does not throw an error for  __stack_chk_fail_local with the exact same arguments for the gcc command like on Buildbot according to config.log?
I would really like to reproduce the error so I may be able to deliver clean solutions.

sbyx, 2015-07-05:
Check your generic build settings and make sure kernel and user space stack smashinh protection is on.

tru7, 2015-07-05:
Both are set to Regular.