openwrt/packages #1553: ddns-scripts error in lucy web interface

finasfinas, 2015-07-09:
just compiled 	OpenWrt Chaos Calmer r46276 / LuCI Master (git-15.188.37359-6d788d8) from new git pull and there is no dynamic dns menu, but dns placeholder appears on the overview page. when I click there I get this:

No page is registered at '/admin/services/ddns'.
If this url belongs to an extension, make sure it is properly installed.
If the extension was recently installed, try removing the /tmp/luci-indexcache file.

chris5560, 2015-07-09:
Never seen this before, because the code for status/overview is part of luci-app-ddns package.
So all code is there or not.
For me it's not clear if you build with or without luci-app-ddns.
I found that sometimes you need to "make distclean" to be sure that no old stuff is compiled into new image. Also to make sure no old stuff exists when there were changes to build scripts, and there are some the last days.

finasfinas, 2015-07-09:
It was a clean build, started from fresh. Included luci-app-ddns with no-ip.com support.
The only thing that I changed from this build to previous ones is that I selected the minify lua sources option. maybe this is what is breaking it.

chris5560, 2015-07-09:
Hi,

I just setup a fresh build env and did the following compiles
x86 generic uclibc	   - with and without minify lua
x86 generic default (musl) - with and without minify lua

On my Virtualbox environment everything works fine (as soon as the wan connection is up and running)
The wan connection is needed because the ddns-scripts are using nslookup to verify addresses and luci-app-ddns uses ddns-scripts.
Without wan connection nslookup takes a while to timeout and this will force lua to wait until the system call comes back with an answer.

Sorry no other problem found so far.
Have you checked images available from download.openwrt.org

finasfinas, 2015-07-10:
I found the problem. It happens when I select the "Remove ipkg/opkg status data files in final images" option.

finasfinas, 2015-07-14:
chris, have you had time to look into this?

chris5560, 2015-07-14:
I looked for it. LuCI-app-ddns tries to detect installed ddns-scripts version.
If you remove ipkg/opkg status data this informations are missing.
I need to think about a way to detect installed package versions without using opkg status info.
I think it's not really possible.
To fix for you short term edit /usr/lib/lua/luci/controller/ddns.lua
Inside function index() directly behind local definitions you find: 
```lua
	-- do NOT start it not ddns-scripts version 2.x
	if not verok then
		return
	end
```
Simply remove this if / end - block or comment out with "--" at line start.

finasfinas, 2015-07-14:
but this does not happen with other packages that have a luci web interface, maybe you can see how they work? ( noob sugestion, I'm sure you looked at it already! ) 

thanks!

chris5560, 2015-07-15:
Right, they are not checking version of their "base" program.
I do to prevent users to mix ddns-scripts and luci-app-ddns versions.
luci-app-privoxy and luci-app-radicale also doing version check.
I think in general it make sense because versions of the "base" program might support or not options configurable via webui.

finasfinas, 2015-07-15:
ok, maybe you can implement a tag on the script that outputs it's version and then you run the script to check the output of it's version.

finasfinas, 2015-07-15:
in better english:
maybe you can implement an option on the script to output it's version, like --version or something, and then run the script to check it's version.

chris5560, 2015-07-16:
Already on my todo list but not planed for CC 15.05.
I'm currently updateing DDNS Client Wiki pages and waiting for CC to be released.

diizzyy, 2017-06-03:
@chris5560 
I assume this is resolved?

chris5560, 2017-06-05:
Yes

diizzyy, 2017-06-05:
@thess @hnyman 
Close this close

closed 2017-06-06 by thess