openwrt/packages #1562: wifidog: fails to build with recent cyassl/wolfssl

hauke, 2015-07-12:
wifidog does not compile any more with the cyassl/wolfssl version used in trunk.

```
checking cyassl/ssl.h usability... yes
checking cyassl/ssl.h presence... yes
checking for cyassl/ssl.h... yes
checking for library containing CyaSSLv23_client_method... no
configure: error: unable to find the CyaSSLv23_client_method function.
make[2]: *** [/mnt/dl/slave/brcm47xx/build/build_dir/target-mipsel_mips32_musl-1.1.10/wifidog-tls/wifidog-1.2.1/.configured_yyyyyyyy] Error 1
make[2]: Leaving directory `/mnt/dl/slave/brcm47xx/build/feeds/packages/net/wifidog'
```

hauke, 2015-07-12:
@mhaas

mhaas, 2015-07-25:
Thank you. Looks like wolfSSL completely changed the method names. Great!

mhaas, 2015-07-25:
Wifidog upstream will hopefully release 1.3.0 soon, which will contain this fix: https://github.com/wifidog/wifidog-gateway/pull/225

diizzyy, 2017-06-03:
@thess @hnyman 
Resolved, please close

closed 2017-06-03 by thess