openwrt/packages #1569: Packaging of crelay

ondrej1024, 2015-07-13:
Hi,

I am the developer of the [crelay project](https://github.com/ondrej1024/crelay), a program for controlling USB relay cards. Some time ago I was asked if this software would run on OpenWRT and I don't see why not. The software was developed with small embedded sytems in mind so OpenWRT seem like a natural choice for it.

Could any of the devs here look into building an OpenWRT package for crelay? It shouldn't be a big issue for someone with the proper dev environment. What do you think?

Ondrej

fpopa, 2015-07-13:
If nobody wants to, I could try to maintain it.

Give me one week to set it up.

ondrej1024, 2015-07-13:
That sounds great. I had some discussion about it with a user called @jabss [here](https://github.com/ondrej1024/crelay/issues/3). Let me know if you need any more information from me.

fpopa, 2015-07-13:
Okay, I'll take a look at it later.

fpopa, 2015-07-20:
Hey, sorry for this, but I think I've overestimated my time managment skills?

I didn't managed to do it, I'd say don't count on me, maybe I will try it again and I will come back here only if I succeed.

ondrej1024, 2015-08-01:
OK, don't worry. I appreciate you tried. 
If there is any other volunteer,  It would be great. Building crelay is really straight forward if you already have built other packages for OpenWRT and have the environment set up.

commodo, 2015-08-17:
@ondrej1024 
Hello,

Let's pick this up then.
Package is here: https://github.com/commodo/packages/commits/crelay at this commit: https://github.com/commodo/packages/commit/4966bb48a0598ddaa172f769af1c1d083da2b2b1

I have to commend you for it; it's a fairly straightforward build, and packages really easy/nicely to OpenWRT.
I got to the point where `crelayd` starts up via service and I get (via the webpage)
```
ERROR: No compatible device detected !
```
So, it's ready to be merged in the main official packages.

Looking around other OpenWRT packages seems nobody automatically configures the OpenWRT firewall yet, so I had to disable mine via `/etc/init.d/firewall stop`.
I've added a `postinst` hook so that people who would install this, would get notified that they need to install a firewall rule; maybe later, a sample firewall rule could be added.
The `mailman` package does the same.

Anyway; I'd prefer to not be the maintainer for this package in OpenWRT, since I already have my plate full (with stuff).
If you'd feel like you could take this part from here and handle it, I'd be happy to help along the way.
I think your project would be a great addition to OpenWRT, to make it a part of the whole IoT thing.
Those relays look like they'd be great for doing lots of neat stuff/apps, and that's why I took the time to lay the foundation for this package.

If you don't want to pick this up, then I'll stay as maintainer for the OpenWRT package, but I will add a note that I might have some timeouts/delays to handle updates/communication for this package.

Thanks
Alex

ondrej1024, 2015-08-17:
@commodo 
Hi Alex,
thank you very much for your effort. Maybe you should coordinate your activities with @thess who has also started building a crelay package. For some reason the discussion about it ended up in the crelay issue tracker: https://github.com/ondrej1024/crelay/issues/3#issuecomment-127275750

I got some comments on the program there and I started doing some modifications to improve mainly the configuration. But first of all I rewrote the Conrad relay card handling in *crelay* because this code depended on a patch of the cp210x kernel driver which was really awkward. With the rewrite, I removed this dependency and now the driver is not needed any more. I released the new version 0.8 with this change on Friday.

Myself I don't have any experience with OpenWRT development so I think it would be difficult for me to be the maintainer of the package. Therefore I would prefer if one of you OpenWRT experts do this.

Thanks, Ondrej

commodo, 2015-08-17:
That's neat.
Should we close this discussion and move things to the other thread ? :)

commodo, 2015-08-17:
Ah, I thought that other discussion was on the OpenWRT packages repo.
No objections then to keep this open as well.

thess, 2015-08-17:
@commodo - I have a package that I put together a week or so ago and have been working with @ondrej1024 to get the cp210x support resolved. I intend to push my package in a day or 2 when there is a release ready to go. You can see my preliminary submission here: https://github.com/thess/OpenWrt-packages/tree/crelay-import

I was intending to be the package maintainer. I'll close this when I merge the package

PR #1662 (2015-08-18): crelay: import usb relay control daemon
@ondrej1024, @commodo -- Any changes or additions? OK to merge?

Closes #1569
Additional discussion here: https://github.com/ondrej1024/crelay/issues/3
closed 2015-08-18 by thess