openwrt/packages #1585: new collectd 5.5.0 => collectd-mod-disk error build (missing dependencies for the following libraries libudev.so.0)

build000, 2015-07-18:
mkdir -p /magazyn/Pliki/openwrt/trunk/bin/ar71xx/packages /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/collectd-5.5.0/ipkg-ar71xx/collectd-mod-disk/CONTROL /magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/pkginfo
install -d -m0755 /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/collectd-5.5.0/ipkg-ar71xx/collectd-mod-disk/usr/lib/collectd
for m in disk; do cp -fpR /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/collectd-5.5.0/ipkg-install/usr/lib/collectd/${m}.so /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/collectd-5.5.0/ipkg-ar71xx/collectd-mod-disk/usr/lib/collectd/ ; done
find /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/collectd-5.5.0/ipkg-ar71xx/collectd-mod-disk -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package collectd-mod-disk is missing dependencies for the following libraries:
libudev.so.0
Makefile:301: recipe for target '/magazyn/Pliki/openwrt/trunk/bin/ar71xx/packages/packages/collectd-mod-disk_5.5.0-1_ar71xx.ipk' failed
make[3]: *** [/magazyn/Pliki/openwrt/trunk/bin/ar71xx/packages/packages/collectd-mod-disk_5.5.0-1_ar71xx.ipk] Error 1
make[3]: Leaving directory '/magazyn/Pliki/openwrt/trunk/feeds/packages/utils/collectd'
package/Makefile:191: recipe for target 'package/feeds/packages/collectd/compile' failed
make[2]: *** [package/feeds/packages/collectd/compile] Error 2
make[2]: Leaving directory '/magazyn/Pliki/openwrt/trunk'
package/Makefile:188: recipe for target '/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/stamp/.package_compile' failed
make[1]: *** [/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/stamp/.package_compile] Error 2
make[1]: Leaving directory '/magazyn/Pliki/openwrt/trunk'
/magazyn/Pliki/openwrt/trunk/include/toplevel.mk:181: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2

... udev of course build and installed

ClaymorePT, 2015-11-26:
I have the same problem.

find /home/ArkHDD/OpenWRT/trunk/build_dir/target-x86_64_glibc-2.21/collectd-5.5.0/ipkg-x86_64/collectd-mod-disk -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package collectd-mod-disk is missing dependencies for the following libraries:
libudev.so.0
Makefile:302: recipe for target '/home/ArkHDD/OpenWRT/trunk/bin/x86-glibc/packages/packages/collectd-mod-disk_5.5.0-5_x86_64.ipk' failed

FreifunkUFO, 2016-05-09:
I have the same problem.
i was able to disable that plugin in order to use all other collectd stuff normally
..../package/feeds/packages/collectd/Makefile
* move disk-modul from enabled plugin to disabled 
* comment out that line: #$(eval $(call BuildPlugin,disk,disk usage/timing input,disk,))
see also: https://plan.leipzig.freifunk.net/issues/248

dfc-d, 2016-05-09:
On 16-05-09 10:11 AM, FreifunkUFO wrote:
> I have the same problem.
> i was able to disable that plugin in order to use all other collectd
> stuff normally
> ..../package/feeds/packages/collectd/Makefile
>
>   * move disk-modul from enabled plugin to [...]

I send a PR some time back to disable use of libudev with collectd for
this very reason.   Not sure why it hasn't been applied yet.

Regards,

Daniel

FreifunkUFO, 2016-05-09:
how is that done in detail? please add that PR or hints here

closed 2016-05-22 by hnyman