openwrt/packages #1593: latest changes kernel version (4.2.1) => exfat-nofuse build error

build000, 2015-07-19:
(...)
make[3]: Entering directory '/magazyn/Pliki/openwrt/trunk/feeds/packages/kernel/exfat-nofuse'
make -C "/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/linux-4.1.2" ARCH="mips" CROSS_COMPILE="mips-openwrt-linux-musl-" M="/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301" CONFIG_EXFAT_FS=m modules
make[4]: Entering directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/linux-4.1.2'
  CC [M]  /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.o
/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.c:1340:17: error: 'new_sync_read' undeclared here (not in a function)
  .read        = new_sync_read,
                 ^
/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.c:1341:17: error: 'new_sync_write' undeclared here (not in a function)
  .write       = new_sync_write,
                 ^
/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.c: In function 'exfat_direct_IO':
/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.c:1618:6: warning: passing argument 1 of 'blockdev_direct_IO' makes pointer from integer without a cast [enabled by default]
      offset, exfat_get_block);
      ^
In file included from ./arch/mips/include/asm/elf.h:11:0,
                 from include/linux/elf.h:4,
                 from include/linux/module.h:14,
                 from /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.c:49:
include/linux/fs.h:2673:23: note: expected 'struct kiocb *' but argument is of type 'int'
 static inline ssize_t blockdev_direct_IO(struct kiocb *iocb,
                       ^
/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.c:1618:6: warning: passing argument 2 of 'blockdev_direct_IO' from incompatible pointer type [enabled by default]
      offset, exfat_get_block);
      ^
In file included from ./arch/mips/include/asm/elf.h:11:0,
                 from include/linux/elf.h:4,
                 from include/linux/module.h:14,
                 from /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.c:49:
include/linux/fs.h:2673:23: note: expected 'struct inode *' but argument is of type 'struct kiocb *'
 static inline ssize_t blockdev_direct_IO(struct kiocb *iocb,
                       ^
/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.c:1618:6: warning: passing argument 3 of 'blockdev_direct_IO' from incompatible pointer type [enabled by default]
      offset, exfat_get_block);
      ^
In file included from ./arch/mips/include/asm/elf.h:11:0,
                 from include/linux/elf.h:4,
                 from include/linux/module.h:14,
                 from /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.c:49:
include/linux/fs.h:2673:23: note: expected 'struct iov_iter *' but argument is of type 'struct inode *'
 static inline ssize_t blockdev_direct_IO(struct kiocb *iocb,
                       ^
/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.c:1618:6: warning: passing argument 4 of 'blockdev_direct_IO' makes integer from pointer without a cast [enabled by default]
      offset, exfat_get_block);
      ^
In file included from ./arch/mips/include/asm/elf.h:11:0,
                 from include/linux/elf.h:4,
                 from include/linux/module.h:14,
                 from /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.c:49:
include/linux/fs.h:2673:23: note: expected 'loff_t' but argument is of type 'struct iov_iter *'
 static inline ssize_t blockdev_direct_IO(struct kiocb *iocb,
                       ^
/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.c:1618:6: warning: passing argument 5 of 'blockdev_direct_IO' makes pointer from integer without a cast [enabled by default]
      offset, exfat_get_block);
      ^
In file included from ./arch/mips/include/asm/elf.h:11:0,
                 from include/linux/elf.h:4,
                 from include/linux/module.h:14,
                 from /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.c:49:
include/linux/fs.h:2673:23: note: expected 'int (*)(struct inode *, sector_t,  struct buffer_head *, int)' but argument is of type 'loff_t'
 static inline ssize_t blockdev_direct_IO(struct kiocb *iocb,
                       ^
/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.c:1618:6: error: too many arguments to function 'blockdev_direct_IO'
      offset, exfat_get_block);
      ^
In file included from ./arch/mips/include/asm/elf.h:11:0,
                 from include/linux/elf.h:4,
                 from include/linux/module.h:14,
                 from /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.c:49:
include/linux/fs.h:2673:23: note: declared here
 static inline ssize_t blockdev_direct_IO(struct kiocb *iocb,
                       ^
/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.c: At top level:
/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.c:1674:2: warning: initialization from incompatible pointer type [enabled by default]
  .direct_IO   = exfat_direct_IO,
  ^
/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.c:1674:2: warning: (near initialization for 'exfat_aops.direct_IO') [enabled by default]
scripts/Makefile.build:258: recipe for target '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.o' failed
make[5]: *** [/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/exfat_super.o] Error 1
Makefile:1383: recipe for target '_module_/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301' failed
make[4]: *** [_module_/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301] Error 2
make[4]: Leaving directory '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/linux-4.1.2'
Makefile:51: recipe for target '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/.built' failed
make[3]: *** [/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_nand/exfat-nofuse-git-20150301/.built] Error 2
make[3]: Leaving directory '/magazyn/Pliki/openwrt/trunk/feeds/packages/kernel/exfat-nofuse'
package/Makefile:191: recipe for target 'package/feeds/packages/exfat-nofuse/compile' failed
make[2]: *** [package/feeds/packages/exfat-nofuse/compile] Error 2
make[2]: Leaving directory '/magazyn/Pliki/openwrt/trunk'
package/Makefile:188: recipe for target '/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/stamp/.package_compile' failed
make[1]: *** [/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/stamp/.package_compile] Error 2
make[1]: Leaving directory '/magazyn/Pliki/openwrt/trunk'
/magazyn/Pliki/openwrt/trunk/include/toplevel.mk:181: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2

KanjiMonster, 2015-08-20:
@br101

build000, 2015-09-01:
https://github.com/openwrt/packages/commit/7d12c8be58695e4ca4d40d8f2e6ce2e57720c88c - fix problem
confirm - builds and working properly.

closed 2015-09-01 by build000