openwrt/packages #1602: dnscrypt-proxy 1.6.0

freegigi, 2015-07-21:
dnscrypt-proxy is two versions behind. the latest release is 1.6.0.
i tried using 1.4.3, but it kept crashing. 1.6.0 on a trunk build works fine.

Thank you.

KanjiMonster, 2015-08-20:
@damianorenfer

damianorenfer, 2015-08-21:
Thanks for reporting. Could you provide more information about the crash please ? 

It is not the lastest version, but as I'm not an OpenWrt expert I don't know the versionning workflow... Should I update the version in for-15.05 ? Or does someone else syncronize with master branch ?

damianorenfer, 2015-08-21:
Found an answer CONTRIBUTINB.md : 
>Do NOT add new packages and do NOT do major upgrades of packages here

damianorenfer, 2015-08-27:
@sbyx or @thess : Is there a way to update to 1.6 in 15.05 branch ?

Thanks

thess, 2015-08-27:
Submit a PR on the for-15.05 branch. Include [for-15.05] in the request title. Give an explanation for the necessity to do the upgrade.

But first... Test the fix in the 15.05 environment

rugk, 2016-04-03:
:+1: for updating. See: https://forum.openwrt.org/viewtopic.php?pid=317892

**Why update to 1.5.0?**
Because v1.50 dnscrypt supports ephemeral keys, which config option is already listed [in the wiki](https://wiki.openwrt.org/inbox/dnscrypt).

**Why update to v1.6.1?**
Because according to the [release notes](https://github.com/jedisct1/dnscrypt-proxy/releases/tag/1.6.1) a vulnerability was fixed there...

thess, 2017-04-08:
Closing this - over a year old

closed 2017-04-08 by thess