openwrt/packages #1620: For-15.05: Snort's /etc/init.d/snort does not read /etc/config/snort's config_file?

MikePetullo, 2015-07-27:
My router's /etc/snort/snort.conf contains:

    config snort 'snort'
    	option config_file '/etc/snort/snort.conf'
    	option interface 'eth1'

However, it does not seem that /etc/init.d/snort honors the config_file option. To get snort to work, I had to add "-c "$config_file" to the "procd_set_param command" line in /etc/init.d/snort.

Should we modify the snort package so that its /etc/init.d/snort makes use of the "-c" argument, or am I missing something?

I am using the for-15.05 branch of the packages feed.

s0m3f00l, 2016-03-02:
How did you get this to work I tried adding the "-c" "$config_file" to the "procd_set_param command" line and it just crashes over and over again.

dl12345, 2016-11-08:
The existing snort init.d script is a little limited. I wrote a new one to allow multiple instances and do a little more error checking

```
#!/bin/sh /etc/rc.common
# Copyright (C) 2015 OpenWrt.org

. /lib/functions/network.sh

START=99
STOP=10

USE_PROCD=1
PROG=/usr/bin/snort
CONFIG=snort
PID_PATH="/var/run"
LOGGER="logger -s -t snort -p authpriv.info"
DELAY=10
TIMEOUT=120

logmessage ()
{
	${LOGGER} "$@"
}

validate_snort_section() 
{

	uci_validate_section ${CONFIG} snort "${1}" \
			'config_file:string' \
			'interface:string' \
			'arguments:string:'

}

wait_on_interface_up() 
{

	local interface=${1}
	local delay=${2}
	local timeout=${3}
	local count=0

	# If using a pppoe or non-ethernet interface, failure to flush the cache will
	# cause the network functions to read from cache, which will not be updated
	# when the interface comes up, leading to spurious failure

	network_flush_cache
	while  ! network_is_up "${interface}" ; do
		sleep "${delay}"
		count="$((${count} + ${delay}))"
		[ "${count}" -ge "${timeout}" ] && return 1;
		network_flush_cache
	done
	return 0
}

start_snort() 
{
	local interface
	local device
	local config_file
	local arguments

	config_get interface "${1}" interface
	config_get config_file "${1}" config_file
	config_get arguments "${1}" arguments

	validate_snort_section ||  {

		logmessage "Syntax error in config file ${CONFIG}"
		return 1

	}

	[ -f "${config_file}" ] || {

		logmessage "Missing config file ${config_file}"
		return 1

	}

	# check for a valid interface

	[ -z "${interface}" ] && {
		logmessage "Unspecified interface"
		return 1
	}

	# fail if the interface doesn't come up after TIMEOUT seconds

	wait_on_interface_up "${interface}" "${DELAY}" "${TIMEOUT}"
	[ "$?" -ne "0" ] && {
		logmessage "Aborting launch of snort due to timeout on ifup of ${interface}";
		return 1
	}

	network_get_device device "${interface}"
        [[ -z "${device}" ]] && {

		logmessage "Unable to locate physical device name for logical interface ${2}"
		return 1

	}

	# remove stale pid files left over after unclean shutdown

	PID_FILE="${PID_PATH}/snort_${device}.pid"

	[ -f "${PID_FILE}" ] && {

		PID=$(cat ${PID_FILE})
		[ -f "/proc/${PID}/cmdline" ] && grep -q snort "/proc/${PID}/cmdline"  && return 0
		rm -f ${PID_FILE}

	}

	procd_open_instance
	procd_set_param command $PROG "-q" "-c" "$config_file" "--daq-dir" "/usr/lib/daq/" \
							"-i" "$device" "-s" "-N" \
							"$arguments" \
							"--pid-path=${PID_PATH}" \
							"--nolock-pidfile"
	procd_set_param respawn
	procd_close_instance

}

start_service() 
{

	config_load 'snort'
	config_foreach start_snort 'snort'

}

service_triggers()
{
	procd_add_reload_trigger "network"
}

restart() 
{
	stop_service
	start_service
}

sighup_snort() 
{

	local interface
	local device

	config_get interface "${1}" interface
	network_get_device device "${interface}"

    [[ -z "${device}" ]] && {

		logmessage "Unable to locate physical device name for logical interface ${2}"
		return 1

	}

	PID_FILE="${PID_PATH}/snort_${device}.pid"

	[ -f "$PID_FILE" ] && {

		PID=$(cat ${PID_FILE})
		[ -f "/proc/${PID}/cmdline" ] && grep -q snort "/proc/${PID}/cmdline" && {
			/usr/bin/kill -s SIGHUP ${PID}
		}
	}
}

reload_service() 
{

	config_load 'snort'
	config_foreach sighup_snort 'snort'

}
```

Example /etc/config/snort

```
config snort 'wan'
	option interface 'wan'
	option config_file '/etc/snort/snort-wan.conf'
	option arguments "-p"

config snort 'lan'
	option interface 'lan'
	option config_file '/etc/snort/snort-lan.conf'
```

diizzyy, 2017-06-07:
@thess @hnyman 
15.05 is frozen and unsupported, close this report.
If this is still an issue, just make a new report.

MikePetullo, 2017-06-07:
this issue appears to remain in master, but I have not been experimenting with Snort for a while. I will open a new bug if I find the time to work with Snort this summer.

closed 2017-06-07 by MikePetullo