openwrt/packages #1621: boost: fails to build on mpc85xx

sairon, 2015-07-27:
It seems that boost is not available for mpc85xx architecture, while it's available for other ones. After some searching, I found out that problem might be the same as reported in Buildroot few months ago: https://patchwork.ozlabs.org/patch/472878/

@ClaymorePT, can you decide what to do - whether to disable failing libraries for mpc85xx or to include the proposed patch?

Also, can anyone tell me (maybe @jow-, @sbyx or @thess), where can I find what package builds are currently failing? It seems to me that buildbot broken packages page (e.g. http://buildbot.openwrt.org:8010/broken_packages/mpc85xx/) contains only results for openwrt trunk + latest packages, where boost builds fine, probably because there's musl instead of uClibc.

ClaymorePT, 2015-07-27:
I will look into it ASAP.

ClaymorePT, 2015-07-27:
BTW, did you try to compile Boost.Log using the regular glibc for that
target?
The report mentions a build attempt using uclibc. Does it build when using
glibc?
I would hate to disable that target due to incompatibility with uclibc.

sairon, 2015-07-27:
I did not - but according to all the clues (OK with musl, OK on other architectures, plus the linked issue), it should be OK with anything else than the specific combination of uClibc + PPC target. The patch sent by Romain Naour fixes the issue but I can't decide whether it is an acceptable solution or just a hack.

You may also disable the problematic library only on the specific target but it'd require some refactoring of DefineBoostLibrary macro (because it does add all the libraries to the boost metapackage anyway).

ClaymorePT, 2015-07-28:
I'm more inclined to disable the package for that specific combination (target + uclibc).

That patch messes with the code itself so i'm reluctant in accepting it without the approval of the Boost devs. Since uclibc is an outdated lib and no longer maintained, it should be avoided whenever possible.

ClaymorePT, 2015-07-30:
FYY
https://github.com/openwrt/packages/pull/1628

sairon, 2015-07-30:
Great, thanks. It think it should be also backported to for-15.05 branch to make boost available on mpc85xx.

And I'm still hoping to receive an answer for my question if/where I can check for failing builds of CC packages.

sbyx, 2015-07-30:
@sairon release builds are done manually and usually someone will come around and open a ticket if a package is failing.

sairon, 2015-07-30:
Thanks @sbyx, so it seems that boost on mpc85xx dropped off the radar...

ClaymorePT, 2015-07-30:
Yes @sairon. Its a matter of bug reporting, when someone hits a problem. In this specific case, it was the combination of target and uclibc.
I myself, don't use uclibc any more, due to incompatibilities with Python. Since the embedded devices to which I develop, have more than 100MB in storage, I opted to use glibc.

ClaymorePT, 2015-11-19:
FYY
https://github.com/openwrt/packages/pull/1991

ClaymorePT, 2016-11-13:
@thess this issue can be closed. Can you do it please?

thess, 2016-11-13:
@sairon - Issue is over 1 yr old and uClibc no longer supported. Closing this.

If there is still any issues with the current version and target support, please open a new issue. Better yet, submit a new PR to solve the problem.

closed 2016-11-13 by thess