openwrt/packages #1631: Host python build miscompiling pyexpat, breaking e.g. glib2 build

paulfertser, 2015-08-01:
Hello,

When python is selected in config, it's built for both host and the target, and the host version overwrites the symlink in `staging_dir/host/bin/` . This becomes problematic for glib2 as it uses pyexpat as part of its gio tests and OpenWrt's host python lacks pyexpat, see below why.

For both host and target python builds the in-tree copy of libexpat is used but for the host build it's not used properly, see the resulting command:

    gcc -pthread -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fno-strict-aliasing -O2 -I/home/paul/openwrt-trunk/staging_dir/host/include -I/home/paul/openwrt-trunk/staging_dir/host/usr/include -I/usr/include/ncursesw -I/usr/include/ncurses -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -DHAVE_EXPAT_CONFIG_H=1 -DUSE_PYEXPAT_CAPI -I/home/paul/openwrt-trunk/build_dir/host/Python-2.7.10/./Modules/expat -I. -IInclude -I./Include -I/home/paul/openwrt-trunk/staging_dir/host/include -I/usr/include/x86_64-linux-gnu -I/usr/local/include -I/home/paul/openwrt-trunk/build_dir/host/Python-2.7.10/Include -I/home/paul/openwrt-trunk/build_dir/host/Python-2.7.10 -c /home/paul/openwrt-trunk/build_dir/host/Python-2.7.10/Modules/pyexpat.c -o build/temp.linux-x86_64-2.7/home/paul/openwrt-trunk/build_dir/host/Python-2.7.10/Modules/pyexpat.o

Due to the way how OpenWrt arranges host builds, `-I/home/paul/openwrt-trunk/staging_dir/host/include` appears earlier in the include path than `-I/home/paul/openwrt-trunk/build_dir/host/Python-2.7.10/./Modules/expat` and as the result pyexpat.o expects unprefixed function names while every object in Modules/expat exports python-namespaced function names. It's discovered only later when importing the module fails with `undefined symbol: XML_SetCommentHandler` which just happened to be the first function referenced. The build continues, just the resulting python lacks pyexpat.

Please feel free to contact me on IRC to discuss this in a more detailed way, if needed.

paulfertser, 2015-08-01:
@commodo , I hope this is the right method to notify you of the issue, TIA.

commodo, 2015-08-01:
Hello,

( This is the right method for notify-ing me )
I usually get all notifications from this repo as I am interested in other packages as well.

I'll look into this in the next couple of days.
How pressing is this issue for you ?
I can try to speed it a bit, but I have a few other issues as well.

Thanks

paulfertser, 2015-08-01:
No pressing at all, so there's no hurry (build issues with the other packages can be easily workarounded by changing the symlink back to the OS's python). Please feel free to ping me if there's anything unclear in the issue description, and TIA.

build000, 2015-08-01:
https://github.com/openwrt/packages/issues/1557 ???

commodo, 2015-09-14:
this can be closed

paulfertser, 2015-09-15:
I can't tell if the fix was correct or not, I'd expect the package maintainer to decide. I see the host configure build flags were changed to use external libexpat but I do not see any depends on that, that doesn't look right to me.

paulfertser, 2015-09-15:
I'm also not sure that this workaround is a good way to handle the issue because it's clearly a bug either in the way OpenWrt does host builds or the way Python compiles libexpat. It's just that using internal libexpat exposes this bug and using external doesn't. But it might surface with some other package some day (if it's a bug in the OpenWrt build system).

commodo, 2015-09-16:
Well, every-once in a while I think about removing the host build of Python 2, since it's a per-requisite of OpenWRT's build-system.
Usually the arguments I find of keeping the host build are:
- it's the same Python on all hosts (if we build it) ; this is not very convincing since 
- we can install various packages on the host Python (that's built) which can allow for various building other packages ; but only pip and setuptools are used for that, and only on seafile-seahub

But the most convincing one, is the effort it takes to do this , which means I have a bit of lazyness in me, with which I have to fight to get things done.

So, I guess (since Python is not that used in the packages feeds builds anyway) we can try to remove the host Python build and try to install Python packages within the build system and use those.
I think for most cases, Python should be consistent enough across most versions ( 2.6 and more recent ) to work fine ; and I would assume that most people are already building OpenWRT on systems with 2.7.

yousong, 2016-08-11:
I take it that the issue was resolved by the referred pull request.

closed 2016-08-11 by yousong