openwrt/packages #1633: opencv fails to build (buildbot) - probably cmake or toolchain bug

plntyk, 2015-08-02:
saw this locally and on 
buildbot on different arch/compiler versions (4.8,4.6):

musl+gcc4.8:
http://buildbot.openwrt.org:8010/broken_packages/ar71xx/opencv/compile.txt
http://buildbot.openwrt.org:8010/broken_packages/brcm2708/opencv/compile.txt

uclibc+gcc4.6:
http://buildbot.openwrt.org:8010/broken_packages/octeon/opencv/compile.txt

```

arm-openwrt-linux-muslgnueabi/lib/crt1.o: In function `_start':
(.text+0x38): undefined reference to `main'

/libc/sysdeps/linux/mips/crt1.S:95: undefined reference to `main'
collect2: ld returned 1 exit status
```

A native build on alpine linux with 
musl libc, gcc 4.9.2 and cmake 3.2.2 is successful

plntyk, 2015-08-03:
building OpenWrt with a tools/cmake 3.3.0 results in the same error

plntyk, 2015-08-03:
http://paste.debian.net/288353/ non-working vs. working

KanjiMonster, 2015-08-20:
@dmrlsn

PR #1696 (2015-09-15): opencv: disable precompiled headers
this commit fixes musl build

precompiled headers are not strictly necessary according to
https://gcc.gnu.org/onlinedocs/gcc/Precompiled-Headers.html

precompiled headers will probably be broken anyway in cross-compile cases

this should fix #1633
closed 2015-09-15 by sbyx