openwrt/packages #1636: no working init script for coova-chilli

gareth41, 2015-08-02:
This was mentioned by someone else too. https://github.com/openwrt/packages/commit/45e0e2a0b96c1f0268a71d50d8ead2d109cfa113

The included init is for the old chillispot not for coova-chilli, hence doesn't work.  The config file is also from the old chillispot.

I compiled the latest trunk from github and also modified the coova-chilli Makefile to grab the latest commit for coova-chilli also on github, then replaced the included init with the
init from here:
https://dev.openwrt.org/attachment/ticket/9995/chilli.patch

This somewhat fixed the problem if running a single instance of chilli, however the script needed some editing to run multiple instances of chilli which also worked when started manually however on reboot caused the router to go into an infinite boot loop requiring a reset or tftp reflash. I'm using a tp-link tl-wr1043nd v2.

teslamint, 2015-08-10:
> The included init is for the old chillispot not for coova-chilli, hence doesn't work. The config file is also from the old chillispot.

No, the init file is taken from patch request by the owner of coova-chilli project(see #1042) and config file is almost same format can found in luci-app-coovachilli(which is broken now) and each options can be found in `chilli --help`.

I don't understand why you and that user thinks that config and init it from Chillispot, but it should work with the current coova-chilli package.

> This somewhat fixed the problem if running a single instance of chilli, however the script needed some editing to run multiple instances of chilli which also worked when started manually however on [...]

It would be easier to make and running multiple chilli instance by creating multiple `config chilli` sections in `/etc/config/chilli` like this:
```
config chilli
    option disabled '0'
    option tundev 'tun0'
    option network 'wlan0-1'
    option uamserver 'https://portal.example.com/for-ssid1'
    option radiusserver1 'radius01.example.com'
    # ... and so on

config chilli
    option disabled '0'
    option tundev 'tun1'
    option network 'wlan0-2' # assume you have created 2nd SSID
    option uamserver 'https://portal.example.com/for-ssid2'
    option radiusserver1 'radius01.example.com'
    # ... and so on
```

diizzyy, 2017-06-07:
Still an issue?

dibdot, 2018-08-17:
no input since 08/2015 - close this by now.

closed 2018-08-17 by dibdot